pypilot/pypilot #3: Calibration error

sailoog, 2017-10-08:
I am getting this error when calibrating with MPU-9255

len 2
len 2
len 3
len 4
len 5
len 6
FitPoints 6
sphere bias fit [15.891722827491156, 13.55440349266639, 22.169804683109746, 30]
sphere fit [13.732123991027981, 14.216547016259632, -7.1043097324847055, 53.27526643071052]
new bias fit [13.982396574152586, 14.2876672366296, -5.1514629021533693, 51.48872485599539, 0.905202140279608] 64.8505217286
fit [[13.982396574152586, 14.2876672366296, -5.1514629021533693, 51.48872485599539, 0.905202140279608], [15.891722827491156, 13.55440349266639, 22.169804683109746, 30], [13.732123991027981, 14.216547016259632, -7.1043097324847055, 53.27526643071052], [13.732123991027981, 14.216547016259632, -7.1043097324847055, 53.27526643071052, 1, 1]]
bias update only
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.1-py2.7-linux-armv7l.egg/pypilot/calibration_fit.py", line 314, in CalibrationProcess
    print 'coverage', coverage, 'new fit:', fit, 'sphere bias difference', sbd
UnboundLocalError: local variable 'sbd' referenced before assignment
coverage 177.673247975 new fit: [[15.891722827491156, 13.55440349266639, 22.169804683109746, 30], [15.891722827491156, 13.55440349266639, 22.169804683109746, 30], [13.732123991027981, 14.216547016259632, -7.1043097324847055, 53.27526643071052], [13.732123991027981, 14.216547016259632, -7.1043097324847055, 53.27526643071052, 1, 1]] sphere bias differencepipe full (1) points pipe[1] cannot send
pipe full (10) points pipe[1] cannot send
pipe full (100) points pipe[1] cannot send

seandepagnier, 2017-10-08:
sorry about this.  Please check out the older revision.

I am reworking the calibration hopefully making it more accurate and smarter.

sailoog, 2017-10-12:
ok, back to working version, thanks.

closed 2017-10-12 by sailoog