pypilot/pypilot #6: Doesn't compile for Windows - Json

rgleason, 2018-06-15:
"C:\Users\Rick\Documents\GitHub\pypilot_pi\build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\Users\Rick\Documents\GitHub\pypilot_pi\build\pypilot_pi.vcxproj" (default target) (3) ->
(ClCompile target) ->
  C:\Users\Rick\Documents\GitHub\pypilot_pi\src\jsoncpp\json_value.cpp(665): error C2668: 'Json::va
lueToString' : ambiguous call to overloaded function [C:\Users\Rick\Documents\GitHub\pypilot_pi\bui
ld\pypilot_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\pypilot_pi\src\jsoncpp\json_writer.cpp(382): error C2668: 'Json::v
alueToString' : ambiguous call to overloaded function [C:\Users\Rick\Documents\GitHub\pypilot_pi\bu
ild\pypilot_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\pypilot_pi\src\jsoncpp\json_writer.cpp(452): error C2668: 'Json::v
alueToString' : ambiguous call to overloaded function [C:\Users\Rick\Documents\GitHub\pypilot_pi\bu
ild\pypilot_pi.vcxproj]
  C:\Users\Rick\Documents\GitHub\pypilot_pi\src\jsoncpp\json_writer.cpp(670): error C2668: 'Json::v
alueToString' : ambiguous call to overloaded function [C:\Users\Rick\Documents\GitHub\pypilot_pi\bu
ild\pypilot_pi.vcxproj]

    8 Warning(s)
    4 Error(s)

seandepagnier, 2018-06-16:
It needs stdc++11 somehow, not sure how to fix it for msvc

seandepagnier, 2018-06-18:
this is actually in the wrong repository, should be in pypilot_pi not pypilot

closed 2018-06-18 by seandepagnier