pypilot/pypilot #12: void read_anenometer() issue

dmdelorme, 2018-11-27:
I am having trouble getting proper lpdir...

```
`read_anenometer(){``
> 
>   ` //Serial.println(lpdir); //good expected value
  if (dt >= 100) { // output every 100ms  
    last_time += 100;
    //Serial.println(lpdir); //bad`strange value
.......
  }
//Serial.println(lpdir); //good expected value
}
```
the reading for speed is fine but the direction is all messed up in the output loop I am confused?

seandepagnier, 2018-11-27:
are you sure the analog pin is receiving a sensible voltage for the
wind direction?

Sean

dmdelorme, 2018-11-27:
yes it is reading proper up to the point that //Serial.println(lpdir); //bad`strange value i e the if loop once it exits the loop i have a proper value?? strange as it is fine before and after. i get a value of 4.41 when i am in the loop regardless of the direction of the windvane. I can not see a reason it would write over the variable in the if statement. The speed is correct in the loop.

seandepagnier, 2018-11-27:
I'm not sure what hardware you are using.

You might need to print values and debug from serial.

All I can say is the code in git with arduino nano and the connections
listed is working for me and others, not sure what you did different.

Sean

seandepagnier, 2018-11-28:
> i found this by printing the values to see at what point it failed.
> I am using a nano with davis wind Instrument. speed d1 direction A4 i was
> getting a fuse error but all the values before and after the if timer loop
> were fine and speed is correct. I change the lines to reflect the change in
> the analog pin
> I have some new nanos coming this week and I will try again and look for a
> typo. Probable a typo on my part somewhere?  I ripped out the lcd and bme280
> stuff and spent days reading though the ADMUX stuff over and over and
> over... Well I do some more work and get back to you on what i fouled up.

You can compare the fuses, it is usually because bod isn't enabled.
Without this, the memory can get corrupted eventually so better to
enable it.

dmdelorme, 2018-11-29:
I got good readings now.. I had taken out all the lcd and bmp280 stuff so at the end in the main loop. I do not call 
```
`#ifdef ANENOMETER
    read_anenometer();
    draw_anenometer();
#else
    draw_barometer_graph();
#endif
```
I just call read_anenometer();
I added delay(1);
```
#ifdef ANENOMETER
  read_anenometer();
  delay(1);
//Serial.println("again");
#else
Serial.println("Anenometer not defind")
#endif`
```
I noticed when i printed the value of lpdir to serial i would get 45 to 46 per every 100 mills the" timing of the if loop" and good readings and a good nmea string in the if loop.
but if i did not i would get 4.41 as lpdir ?
so i added a delay and it worked?
I would like to understand why i needed a delay?
Is there something wrong with this part as i did get 4 X 45 bad values for lpdir before the value was correct?
```
float sensorValue = 0;

  cli();
  uint32_t val = adcval[1];
  uint16_t count = adccount[1];
  // reset the adc
  adcval[0] = adcval[1] = 0;
  adccount[0] = adccount[1] = -4; // skip the first 4 readings after changing channel
  adcchannel = 0;
  ADMUX = _BV(REFS0) | 3; // changed to A4  01000100

```
 I will check the values of val more later but for now I have data going into openplotter.

closed 2018-11-29 by dmdelorme