pypilot/pypilot #15: pypilot_calibration not starting up

W-Geronius, 2019-02-22:
Hello,
I am getting this error when I try to calibrate my moitessier hat sensors (as seen in openplotter debug window) - Mode 'IMU only'

> `failed to load pywavefront: invalid syntax (obj.py, line 403)
Traceback (most recent call last):  

>  File "/usr/local/bin/pypilot_calibration", line 11, in <module>
    load_entry_point('pypilot==0.2', 'console_scripts', 'pypilot_calibration')()  

>  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.2-py2.7-linux-armv7l.egg/ui/autopilot_calibration.py", line 472, in main  CalibrationDialog().ShowModal()  

>  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.2-py2.7-linux-armv7l.egg/ui/autopilot_calibration.py", line 30, in __init__    super(CalibrationDialog, self).__init__(None)  

>  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.2-py2.7-linux-armv7l.egg/ui/autopilot_control_ui.py", line 268, in __init__    self.m_panel4.Layout()  

>  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 11496, in Layout
    return _core_.Window_Layout(*args, **kwargs)
wx._core.PyAssertionError: C++ assertion "m_widget" failed at ../src/gtk/window.cpp(2831) in DoSetSize(): invalid window` 

it is the same error message I see, when starting interactively via command pypilot_calibration

Please advise what to do
Thank you
Michael

seandepagnier, 2019-02-22:
Is there more output text if you scroll up?

W-Geronius, 2019-02-23:
no more output - openplotter continues without opening the calibration window. When called via command line - back to prompt ...

W-Geronius, 2019-02-23:
Hi Sean, I peeked around in the code and got the impression it's a grafics issue caused by my setup. Noone else appears to encounter this problem.

After connecting my pi to a standard HDMI monitor, everything looks fine. So it does, when using standard VNC-session. In virtual [VNC-Sessions ](https://www.realvnc.com/en/connect/docs/raspberry-pi.html#raspberry-pi-virtual)however, the above mentioned errors occur in whatever resolution settings.

From my side it's fine to close this issue as I can live with that restriction. Maybe you want to explore it or add it to the docs.

best,
Michael

seandepagnier, 2019-02-23:
I recently changed this to use the newer pywavefront.  Are you sure no output before:
`failed to load pywavefront: invalid syntax (obj.py, line 403)
Traceback (most recent call last):

?

I want to at least fix it so the imu tab is blank if it fails, not abort

W-Geronius, 2019-02-23:
Yes, nothing before that. The only message before is 'No humidity sensor detected' which is the correct standard message.

sailoog, 2019-02-23:
Does that means that the current pypilot version should use the latest PyWavefront?

pip install PyWavefront

seandepagnier, 2019-02-23:
yes

sailoog, 2019-02-23:
OK, thanks. Working on OpenPlotter v2.x.x now, please let me know anything else I should take into account.

closed 2019-02-27 by seandepagnier

sailoog, 2019-04-07:
Reopening
Now calibration starts but the boat is not rendered in imu tab probably because this:
failed to load pywavefront: invalid syntax (obj.py, line 403)

pywavefront uses the command _yield from_ which is not supported in python2.7 only in python3.

seandepagnier, 2019-04-10:
sorry I think you messed my message?

This issue should be fixed now in pip, also see:
https://github.com/pywavefront/PyWavefront/issues/96

sailoog, 2019-04-11:
Fixed in pi-gen too, thanks!