pypilot/pypilot #23: Webapp Basic/Simple/Learning selection not keeping, gains not filtered.

gvaughan67, 2019-04-22:
What I believe is an attempt to filter for the most important Gains and reduce clutter actually results in an even larger list of Gains onscreen (D.basic D. learning etc)

Unless I've misunderstood the purpose of this select list, perhaps choices would better be labeled differently:

quick start, basic, advanced?

Or could use "more" and "less" to expand the list.

seandepagnier, 2019-04-23:
I think the webapp lags behind the other interfaces for multiple
pilots.  Please be patient.

gvaughan67, 2019-04-26:
Most of this problem was resolved by installing python-gevent-websocket 
the gain selections aren't limited by the dropdown selection of learning, (cosmetic really) but selections are kept.

seandepagnier, 2019-05-21:
Eventually will fix better.