pypilot/pypilot #25: boat not rendered in calibration window

sailoog, 2019-08-26:
Running Raspbian Buster in RPi4 I get this error:

failed to load pywavefront: glXSwapIntervalMESA is not exported by the available OpenGL driver.  MESA_swap_control is required for this functionality.

OpenGL is enabled by default in RPi4.

I have found this, it may help:
https://www.mail-archive.com/pyglet-users@googlegroups.com/msg11142.html

sailoog, 2019-08-29:
I managed to install pypilot as python3 and I had to make a lot of changes in code. To be python3 compatible most imported packages that are in the same folder have to be called this way: 
`from .autopilot import *`
instead of
`import autopilot`
and then make changes in code when the packages are called removing autopilot.
Dependencies have to be installed for python3 (pywavefront, pyglet, python3-opengl ...) and RTIMULib compiled for python3. Finally it works but the boat is not rendered either:

`failed to load pywavefront: glXSwapIntervalMESA is not exported by the available OpenGL driver.  MESA_swap_control is required for this functionality.
/usr/local/lib/python3.7/dist-packages/pypilot-0.9-py3.7-linux-armv7l.egg/ui/autopilot_control_ui.py:284: wxPyDeprecationWarning: Call to deprecated item. Use SetSizeHints instead.
  self.SetSizeHintsSz( wx.DefaultSize, wx.DefaultSize )

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.9-py3.7-linux-armv7l.egg/ui/autopilot_calibration.py", line 446, in onSizeGLBoatPlot
    self.boat_plot.reshape(event.GetSize().x, event.GetSize().y)
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.9-py3.7-linux-armv7l.egg/ui/boatplot.py", line 184, in reshape
    glFrustum(-ar, ar, -0.5, 0.5, 2.0, 300.0)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1281,
	description = b'invalid value',
	baseOperation = glFrustum,
	cArguments = (-0.0, 0.0, -0.5, 0.5, 2.0, 300.0)
)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.9-py3.7-linux-armv7l.egg/ui/autopilot_calibration.py", line 446, in onSizeGLBoatPlot
    self.boat_plot.reshape(event.GetSize().x, event.GetSize().y)
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.9-py3.7-linux-armv7l.egg/ui/boatplot.py", line 184, in reshape
    glFrustum(-ar, ar, -0.5, 0.5, 2.0, 300.0)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1281,
	description = b'invalid value',
	baseOperation = glFrustum,
`
I am migrating pypilot to openplotter 2 right now and it is time to decide to migrate to python3 or not. What are your plans?

gaultierq, 2019-09-15:
On OP 1.2.0 alpha, pressing Calibration button in the pypilot tab gives :
```
failed to load pywavefront: pyglet requires a X server with GLX 
[...] 
OpenGL GLx extension is not supported by display
```
running `python ui/autopilot_calibration.py` gives the same result
and when `python3 ui/autopilot_calibration.py`, I get `No module called 'wx'`

seandepagnier, 2019-11-29:
It should all work now with python3.

closed 2019-11-29 by seandepagnier

orca-hydromancer, 2021-09-28:
FYI:
Tested with all latest on RPi4 (py3 ofc).
The calibration rendering doesn't work if I `pip install pyglet`.
If I uninstall pyglet the rendering works.
If I install pyglet from latest master with `pip install --upgrade --user https://github.com/pyglet/pyglet/archive/master.zip` the rendering works.