pypilot/pypilot #30: MPU-9250 not detected

gaultierq, 2019-09-15:
Running `python pypilot/boatimu.py` with a MPU-9250 plugged gives me _No IMU detected_, even though `sudo i2cdetect -y 1` finds it on 0x68.
Moreover the output of 
`watch -n.1 python gyro.py`
gives me some confidence that the MPU is healthy.

[gyro.py](https://tutorials-raspberrypi.com/measuring-rotation-and-acceleration-raspberry-pi/)

seandepagnier, 2019-09-16:
What version of rtimulib2 are you using.

gaultierq, 2019-09-17:
I tried many, but mainly https://github.com/seandepagnier/RTIMULib2/

seandepagnier, 2019-09-18:
There is an example program in RTIMULib2/Linux/Drive

This tests only rtimulib2 and uses no python.   Make sure to delete
the .ini file before running.

Are you sure you don't have a system RTIMULib installed somehow?  Can
you try tinypilot image on your raspberry pi?

sailoog, 2019-09-18:
If you are in Raspbian you may have already installed rtimulib from another package, try:
`sudo apt autoremove sense-hat`

gaultierq, 2019-09-23:
Apparently `sense-hat` was not installed. 
I installed a fresh openplotter_v1.0.0.img and connected my MPU-9250/6500
trying `RTIMULibDrive` from /home/pi gave the same `No IMU found`
The python is working.

seandepagnier, 2019-09-23:
What is the output?  What does i2cdetect say?

gaultierq, 2019-09-24:
```
pi@openplotter:~ $ RTIMULibDrive
Settings file not found. Using defaults and creating settings file
No IMU detected
Using fusion algorithm RTQF
No IMU found
```

```
pi@openplotter:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

seandepagnier, 2019-09-24:
It must be something in the code that considers your mpu9255 not to be
detected.  At this point I would get a working  mpu9255, or single
step the rtimulib2 code to find out why it   doesn't  detect  the
device  on 0x68.

gaultierq, 2019-09-25:
Logging the [result](https://github.com/seandepagnier/RTIMULib2/blob/b4006cacdeae19899df2396c81d9f1d4ac3a8ba1/RTIMULib/RTIMUSettings.cpp#L84) value gives: 0x78 instead of 0x71 or 0x73

closed 2019-11-29 by seandepagnier