pypilot/pypilot #33: server crashing at start

partyvi, 2019-11-11:


partyvi, 2019-11-11:
Server crashes at start - back trace here
Checked on fresh clonned and rebuilt master version
Looks like tensorflow was not pushed to repo

Settings file RTIMULib.ini loaded
Using fusion algorithm Kalman STATE4
read_sensors_d waiting for signal to exit 
BME280 reading failed: [Errno 121] Remote I/O error
launched pypilot pid 2074
connect failed to localhost:21311
ERROR loading learning.py No module named pilots.learning ,  No module named tensorflow
Traceback (most recent call last):
  File "/usr/local/bin/pypilot", line 11, in <module>
    load_entry_point('pypilot==0.10', 'console_scripts', 'pypilot')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.10-py2.7-linux-armv7l.egg/pypilot/autopilot.py", line 183, in <module>
    import pilots
  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.10-py2.7-linux-armv7l.egg/pypilot/pilots/__init__.py", line 19, in <module>
    default += [simple.SimplePilot, basic.BasicPilot, learning.LearningPilot, wind.WindPilot, absolute.AbsolutePilot]
NameError: name 'learning' is not defined
pypilot exited
stopping pypilot pid: 2074
exception stopping pypilot [Errno 3] No such process
pypilot thread exiting

seandepagnier, 2019-11-12:
I have reimplemented the pilot loading again, I hope it works now.

partyvi, 2019-11-13:
Works now -thanks

closed 2019-11-13 by partyvi