pypilot/pypilot #47: Boat not rendered in calibration window

sailoog, 2019-12-28:
In RPi 4 opengl is enabled by default and I am getting this error in calibration window. pypilot is installed as python3 and pywavefront with pip3.
Some users reported that the boat is rendered ok in RPi 3 where opengl is not enabled by default. Not tested.

```
failed to load pywavefront: expected string or bytes-like object

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/ui/autopilot_calibration.py", line 381, in onSizeGLBoatPlot
    self.boat_plot.reshape(event.GetSize().x, event.GetSize().y)
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/ui/boatplot.py", line 184, in reshape
    glFrustum(-ar, ar, -0.5, 0.5, 2.0, 300.0)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1281,
	description = b'invalid value',
	baseOperation = glFrustum,
	cArguments = (-0.0, 0.0, -0.5, 0.5, 2.0, 300.0)
)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/ui/autopilot_calibration.py", line 381, in onSizeGLBoatPlot
    self.boat_plot.reshape(event.GetSize().x, event.GetSize().y)
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/ui/boatplot.py", line 184, in reshape
    glFrustum(-ar, ar, -0.5, 0.5, 2.0, 300.0)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1281,
	description = b'invalid value',
	baseOperation = glFrustum,
	cArguments = (-0.0, 0.0, -0.5, 0.5, 2.0, 300.0)
)
```

sailoog, 2020-01-09:
Update

After applying latest commits this is the message now:

```
failed to load pywavefront: 'MissingFunctionException' object has no attribute 'message'

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/ui/autopilot_calibration.py", line 384, in onSizeGLBoatPlot
    self.boat_plot.reshape(event.GetSize().x, event.GetSize().y)
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/ui/boatplot.py", line 185, in reshape
    glFrustum(-ar, ar, -0.5, 0.5, 2.0, 300.0)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1281,
	description = b'invalid value',
	baseOperation = glFrustum,
	cArguments = (-0.0, 0.0, -0.5, 0.5, 2.0, 300.0)
)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/ui/autopilot_calibration.py", line 384, in onSizeGLBoatPlot
    self.boat_plot.reshape(event.GetSize().x, event.GetSize().y)
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/ui/boatplot.py", line 185, in reshape
    glFrustum(-ar, ar, -0.5, 0.5, 2.0, 300.0)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1281,
	description = b'invalid value',
	baseOperation = glFrustum,
	cArguments = (-0.0, 0.0, -0.5, 0.5, 2.0, 300.0)
)

```

seandepagnier, 2020-01-09:
there is a bug in pywavefront with python3 for a missing
implementation of a gl extension which is reported to exist.

sailoog, 2020-02-07:
It seems that latest Raspbian kernel solves this issue.

closed 2020-02-07 by sailoog