pypilot/pypilot #48: No module named 'ugfx'

sailoog, 2019-12-29:
pypilot_lcd sends this error to log and stops. I do not know where to find ugfx, is this something to compile?
pypilot installed as python3 in RPI4.

```
Dec 29 18:26:56 openplotter pypilot_lcd[1198]: have gpio for raspberry pi
Dec 29 18:26:56 openplotter pypilot_lcd[1198]: no lirc available
Dec 29 18:26:56 openplotter pypilot_lcd[1198]: Traceback (most recent call last):
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:   File "/usr/local/bin/pypilot_lcd", line 11, in <module>
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:     load_entry_point('pypilot==0.12', 'console_scripts', 'pypilot_lcd')()
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 489, in load_entry_point
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:     return get_distribution(dist).load_entry_point(group, name)
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:     return ep.load()
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 2443, in load
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:     return self.resolve()
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:   File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 2449, in resolve
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:     module = __import__(self.module_name, fromlist=['__name__'], level=0)
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:   File "/usr/local/lib/python3.7/dist-packages/pypilot-0.12-py3.7-linux-armv7l.egg/lcd/lcd.py", line 41, in <module>
Dec 29 18:26:56 openplotter pypilot_lcd[1198]:     from ugfx import ugfx
Dec 29 18:26:56 openplotter pypilot_lcd[1198]: ModuleNotFoundError: No module named 'ugfx'
Dec 29 18:26:56 openplotter systemd[1]: pypilot_lcd.service: Main process exited, code=exited, status=1/FAILURE
Dec 29 18:26:56 openplotter systemd[1]: pypilot_lcd.service: Failed with result 'exit-code'.
```

seandepagnier, 2020-01-19:
pypilot_lcd changed pypilot_hat to support more than just lcd displays.  
The packaging is broken and inconsistent with python3.   I think you must do:
`rm -rf dist build
python setup.py build
sudo python setup.py install`

without the build step the package won't be found.   I don't think this is a problem with python3, but instead the code may not work currently as an installed script.

seandepagnier, 2020-09-25:
this is old and should be fixed

closed 2020-09-25 by seandepagnier