pypilot/pypilot #50: Theming of webclient

FredericGuilbault, 2020-01-20:
Hi, between here and the spring I plan to redo the theme of the webclient so it fit Lysmarine look and feel. 

I can do it on my side as a patch of a fork. 
Or I could put a bit more work in it and PR it as dark theme with a GUI theme switcher or CLI argument.

One one hand I prefer contribution then forks, But on the other hand my need is it have to reflect lysmarine look and feel ( mainly based on adapta-nokto work  https://github.com/adapta-project/adapta-gtk-theme ) so it's not much of a common goal. 

So before I start, would you be interested in integrating such PR or you prefer that I keep it on my side  ? As in a what make more sense for pypilot project.

seandepagnier, 2020-01-20:
I would integrate such a pull request to change the theme.

What do you have in mind?   Could the theme be set via the web
interface?    What would change?  css style or actual layout of the
buttons and control?

Sean

FredericGuilbault, 2020-01-24:
The usual web way to do this is to add a second CSS file that override the base theme when a specific class placed in the body.   Something like `<body class="dark-theme">` 

Then, there is 2 way to make the magic append.
1) The `dark-theme` class class and it's CSS file could be added on the server side uppon CLI argument or config file. 

2) Or the CSS file is always loaded, And the `dark-theme` class is added by a Javascript event bind to a HTML selector. (This might need a cookie to be persistant across reload. I will have to make some research on this to see if it's possible to avoid the cookie.)
 
In all case Ill try to avoid touching the actual code as possible. So layout should pretty much stay the same. Most of the work will append in a new CSS file.

seandepagnier, 2020-01-24:
It would be good if the user could change the theme from the web
interface and store the theme in the browser so different
devices/browsers can be set to different themes.   This is probably
closest to #2.

FredericGuilbault, 2020-02-10:
There is the branch for this feature : https://github.com/FredericGuilbault/pypilot/commits/dark-theme-web

FredericGuilbault, 2020-02-12:
Preview:
Big:
[image]
Small:
[image]
---
Calibration tab
[image]

closed 2020-03-02 by FredericGuilbault

Jamos1988, 2020-03-21:
How would one implement this theme?

FredericGuilbault, 2020-03-21:
@Jamos1988  it's have been merged, so it should be part of pypilot already. 

You need to go on the configuration tab to turn it on.