pypilot/pypilot #51: GUI crash:  'Window_SetSizeHints', expected argument 2 of type 'int'

FredericGuilbault, 2020-02-03:
I realised that gui is not working anymore in lysmarine. 

``
user@lysmarine:~ $ pypilot_control 
Traceback (most recent call last):
  File "/usr/local/bin/pypilot_control", line 11, in <module>
    load_entry_point('pypilot==0.14', 'console_scripts', 'pypilot_control')()
  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.14-py2.7-linux-armv7l.egg/ui/autopilot_control.py", line 397, in main
    AutopilotControl().Show()
  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.14-py2.7-linux-armv7l.egg/ui/autopilot_control.py", line 19, in __init__
    super(AutopilotControl, self).__init__(None)
  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.14-py2.7-linux-armv7l.egg/ui/autopilot_control_ui.py", line 23, in __init__
    self.SetSizeHints( wx.Size( -1,-1 ), wx.DefaultSize )
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk3/wx/_core.py", line 9718, in SetSizeHints
    return _core_.Window_SetSizeHints(*args, **kwargs)
TypeError: in method 'Window_SetSizeHints', expected argument 2 of type 'int'
``
It was working few weeks ago and I didin't touch my build script. 
So I tried to cherry pick differents commits to narrow the problem. 
```
# BAD:    git checkout 1292dd884f8e3c3d7f3073c96aa398cfdfee6b29
# BAD:    git checkout 69702f28423aa7bcdf10ebfa21095d462340bb0d
# BAD:    git checkout 2008b19798a45ba942433305b9b9c61bb8c4686b
# GOOD: git checkout db173ae4409aba2900dfd58c50bf8a409cd954e7
# GOOD: git checkout f25d0770e1f502b7168c1f67e38c4cf1892b541e
```
so commit 2008b19798a45ba942433305b9b9c61bb8c4686b 
is the one that breaks it for me 

I tried with both 
`apt install python-wxgtk3.0` 
and 
`apt install python-wxgtk4.0`

seandepagnier, 2020-02-18:
it works ok with python3, but also I fixed this recently again.

closed 2020-02-18 by seandepagnier

FredericGuilbault, 2020-03-28:
Im confused on how I should make pypilot work under python3.

Dependencies dosn't seem to seem to play well when you mix python2 and python3 and  .deb and pip .

From what I tried it bring more problems then it solve. I took a look at how openplotter do and their package seem to assume a lot of pre-installed things.

Il continue to dig on this....

sailoog, 2020-03-29:
To have pypilot working with python3 you should install these debian dependencies:
https://github.com/openplotter/openplotter-pypilot/blob/e430121f9af7d343efc605cf845fddfd5f0d6233/debian/control#L13
and these pip3 dependencies:
https://github.com/openplotter/openplotter-pypilot/blob/e430121f9af7d343efc605cf845fddfd5f0d6233/openplotterPypilot/pypilotPostInstall.py#L37

FredericGuilbault, 2020-03-29:
Ha! the postInstall. that was the part I missed. Thx @sailoog

FredericGuilbault, 2020-04-04:
With the info from sailoog I been able to make it work :D

@seandepagnier  Now I would like to take what I learned here to improve the Readme file.

But a thing need to be decided. pypilot deps can be found under linux packages or pip packages.

---
I would argue that pip should be prioritized, but that's just my opinion. My points are: 

- pypilot is not shipped as a debian package. so using .deb dependency is not mandatory. 

- pypilot compiler/installer is `setup.py`  from what I read. it support pip dependency checks including version requirement controll. It can automate `pip installs` from  an hard coded list of packages or from a file named `requirements.txt`. This seem to be the python way to do things. 

- Relying on pip is more cross OS then .deb (im thinking of tinycore here).

On the others hand. Some packages are just python binds that rely on C++ libs, So pip packages are shipped with hope that the OS have the requirements installed on the host OS, Debian equivalent have the needed libs flagged as dependencys.

FredericGuilbault, 2020-04-04:
It could look like this : 
```sh
apt-get install -y -q python3 python3-pip libpython3-dev 
apt-get install -y -q swig    # Required by setup.py
apt-get install -y -q libatlas-base-dev  #numpy won't work without this
apt-get install -y -q freeglut3 libgl1 libglu1-mesa libgle3  #Debian package says pyopengl depend on theses 
apt-get install -y -q python-wxgtk4.0 # wxpython is the name of the pip package. But I don't have the balls for this one, too much compiling dependency and compiling. (tests have make my Rpi overheat)
```
Now we can focus on pip
``
pip3 install numpy pillow serial gps pyudev pywavefront pyglet serial scipy gevent-websocket python-socketio flask flask-socketio pyopengl
```