pypilot/pypilot #54: Confusion in supported environements. Around the use of sudo

FredericGuilbault, 2020-02-17:
In the folder `scripts/debian` the .service files assume a user named pi with no-password sudo access. This is clearly raspbian not debian. It's a detail, but I had to debug it.

Alternatively, made this instalation/integration script to fit my situation:  https://github.com/lysmarine/lysmarine_gen/blob/master/lysmarine/32-pypilot/run.sh
You might find some inspiring tweaks in it. 
__

But what consern me it's the fact that I made a quick search across the repo and I found some sudo call in python code. I guess they also assume a no-password sudo access  ??  (as no humans will be there to enter the password ). 

If so, this would mean that running as root or having a user no-password sudo access is mandatory to run pypilot otherwise it's broken ? 

The reason of my questions is that I don't have the full setup to test pypilot and see if it works anyway.  (Cost-less suggestions are also welcome on that side.)

seandepagnier, 2020-02-17:
for now, yes this is an issue.  It won't make pypilot not work but the
process schedule won't be realtime.   There are a few other pieces
that need superuser permissions but nothing critical.

Should the sticky bit be set for certain processes?  A more secure way
would be preferred.

FredericGuilbault, 2020-02-17:
The cheapest change would be at least to limit which process can sudo without root. Like I did in this project: https://github.com/SignalK/set-system-time

The sticky or SGID bit it self won't help here. maybe SUID
This would mean `pypilot.py` would be owned by root. Normal users would be able to launch it and it would run with root privileges. This prevent the user to have enough privileges to do a  `rm -rf /*`  But `pypilot.py` would still have enough privileges to do it. 

ill check deeper in what are exactly sudo launch, Maybe just a question of adding the user to www-data, tty or other groups. maybe there is workaround those commands.

FredericGuilbault, 2020-02-17:
in python code, the calls are :  
```
sudo chrt
sudo renice
sudo chown
```
Im juste brainstorming here, I did not test any of the following. 
### chown ### 
In `pypilot/hat/gpio.py` line 54 : `sudo chown tc /dev/gpiomem` 

Internet say that `/dev/gpiomem` is own by the group name `gpio` Otherwise tutorials show how to create the group `gpio` and make `/dev/gpiomem` part of it.

So if I get right,  you just `usermod -a -G gpio [USER_NAME]` and there is no need to hack the mount point. 

Revelant Examples of what I found : 
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=153654
https://wiki.odroid.com/troubleshooting/gpiomem

### nice ###
The user capability to set nice is limited by this file `/etc/security/limits.conf`
I THINK if you give your user the privilege to nice as they want you won't need root permission to overpass the limits. (now I feel old knowing these things)

In both case, it's typical sysadmin approach. it goes throu system configuration ( and push toward a dedicated pypilot user ). If your target users is packagers like me and openplotter that's good but I woudn't expect lambda user to edit `security/limits.conf`

FredericGuilbault, 2020-02-17:
btw, that is this user named `tc` ?

seandepagnier, 2020-02-18:
pypilot is mostly intended to work as tinypilot which is tinycore linux.   This explains the tc user name and also a lot of other config is simplified compared to debian.

It would be good to remove all references to a specific users or at least use the current username.

As for nice, it needs super user to set negative values.   This should really only be used if chrt does not work or the command isn't found.

The other case of needing superuser is for the webapp if it listens on port 80.   Ports above 1024 don't need it.

It would be nice if pyilot did not need superuser privileges for any process, but had a way to get a superuser process to set the scheduler priorities.

FredericGuilbault, 2020-02-18:
systemd allow variables passed throu the launch of the file. like:  `pypilot@tc.service` or `pypilot@pi.service` 

This could be a way to externalise the user name to the implementation level rather then hard coded.

Would that be a good approach ? 
___

I just took a look at how Im running singalk on port 80 without privielge. 

There is a first systemd service file called  `signalk.socket`. This one open and listen on port 80 with root privileges. 

The second file is `signalk.service` it run signalk  as the `signalk` user with no privilegebut it is able to serve/listen throu the .socket file. 

https://github.com/lysmarine/lysmarine_gen/tree/master/lysmarine/40-signalk/files

FredericGuilbault, 2020-05-20:
I got a second tough about this. And I came to the conclusion that if a software rely on escalated privileges to do it's basic routine actions. it kinda ask mean it should not be run by a user. 

I mean, Running as sudo or as root is pretty close in term of security risk + the Pypilot-core is not  specific to the user session, many users could not run many instances of it. It's pretty much 1 per boat  and it don't need any logged humans to interact with it. (As it's independant GUIs that communicate with it.) 

So the only thing I would need to do is putting /.pypilot/* in /etc/pypilot. I don't see any case where someone would want to run 2 pypilot on the same OS.

seandepagnier, 2021-04-10:
This is still somewhat unresolved.    pypilot only needs root privileges to set two of the processes to realtime priority.   Otherwise it doesn't need privileges of root.   So if it is easier to run as root, or set up as suid root, ok, but it seems like it should change the user to less privilege once the processes are configured.

FredericGuilbault, 2021-04-21:
Does Tiny Core use polkit ? That could be a good way allow the most strict privilege execution

seandepagnier, 2021-04-21:
it only needs to set 2 of the processes to schedule realtime,
otherwise root is not needed.

So perhaps, there is a cleaner way of doing this?  Maybe small script
with suid bit can do this task?

FredericGuilbault, 2021-04-24:
My sysadmin expertise tells me that the modern way to do it is to use `polkit-1`. But it's not universal, some linux don't implement it. 

2nd approach would be to do it in the .service file.  Running pre-exec as root and exec as user. But that make pypilot dependent of it's ,service file. 

The 3rd thing would be to do like Apache and add an argument like `sudo pypilot.py --user=pypilot` So a root process could bootstrap set the nice then do  `su pypilot -c"actualServerExecutable.py"` So the actual server only run with user privilege. 

IDK about suid, applying suid to file that are part of an other package is a big no-no and applying suid to pypilot.py open the door to full root permission for what ever is inside this script file.

seandepagnier, 2021-04-25:
well polkit and service option are probably out for tinypilot, but for openplotter

the third option is probably the best, however it may be possible instead to simply set pypilot with realtime priority when it runs as a service, and then as pypilot creates other processes, it would change them to lower priorities.