pypilot/pypilot #57: Flask web app loses SignalK connection

Jamos1988, 2020-04-16:
Hi Sean,

Was just wondering, what could be possible reasons that the Pypilot Webapp looses it SignalK connection?
Sometimes it can run for days, but sometimes it loses its SignalK connection.

I don't know how to log or troubleshoot it, do you have some advice?
(It is regardless of browser type or device by the way)

Restarting SignalK server, or restarting the sensors from Pypilot tab in OP doesn't work. Only rebooting the Pi fixes it. And very rarely, it just goes away by itself.
In the browser when pressing F12 going to the console it says:

Source map error: Error: NetworkError when attempting to fetch resource.
Resource URL: http://###.###.#.##:3002/static/jquery.min.js
Source Map URL: jquery.min.map

However it also says this when it's working correctly, so I don't think it is relevant.

Also important information, when this issue is present, in SignalK the Pypilot connection is still online. It's only the web application of Pypilot. OpenCPN and Openplotter still operate the autopilot, except for the webapp.

seandepagnier, 2020-04-16:
I am not sure but I have rewritten the connection completely and I'm
currently testing the new branch.   The webapp uses the class which
automatically reconnects now, which it didn't use before.   So it is
likely this issue will go away once the new data format is finally
usable.

Jamos1988, 2020-04-16:
Great to hear that some new coding might fix the problem. Is there a way for me to only upgrade the webapp, and not the underlying pypilot scripts, etc.? I just have it stable now on OP 1.2.0, i know the advice is to go to OP2 but I am not ready for it yet. So I was wondering if I can upgrade only part of Pypilot. Hope it's possible.

seandepagnier, 2020-04-16:
no, it's not possible.   The data format is completely different.

Sean

Jamos1988, 2020-04-16:
Sorry to hear that :(
For the 1.2.0. one, do you have the most recent for that version somewhere? Or could you tell me which one it is?

FredericGuilbault, 2020-05-03:
`jquery.min.map` is for development mainly. It's ok if it's missing

Jamos1988, 2020-06-09:
I would like to troubleshoot this issue some more and need some pointers if possible. Are there some logs that I could monitor in order to know where the disconnection comes from? It would help me so much. Thanks in advance.

seandepagnier, 2020-09-25:
the pypilot_web service has a log output.   I would suggest upgrading pypilot to version 0.22 as there are many changes including the web

seandepagnier, 2020-10-06:
There are now beta images available, could you see if this issue is resolved for you?

Jamos1988, 2020-10-06:
Thanks! Where can I download them?

seandepagnier, 2020-10-06:
pypilot.org/images

Jamos1988, 2020-10-08:
Quick question, will this new image be supporting a pi 3b+ instead of a zero?

seandepagnier, 2020-10-08:
it supports all pi except 4

Jamos1988, 2021-02-03:
Just wanted to mention, the webapp in the newest image from openplotter also does not have a connection:

pypilot server: not connected

I've installed through the openplotter-settings so not through GIT
Any ideas? Is there a log I can look at?

seandepagnier, 2021-04-06:
pypilot_web is a separate service fro pypilot.   So what you indicate suggests that pypilot_web is running, but not pypilot

Jamos1988, 2021-04-06:
Ok thanks for replying, previously when installing op pypilot it would run
the pypilot service also, and I do remember that also in this case. But
I'll check just to be sure. Question, what could prevent the service from
running or how can I make sure it is autostarted?

seandepagnier, 2021-08-07:
I believe the issues are finally resolved with the dependency checking script.

the services on debian systems are typically arranged as systemd scripts

closed 2021-08-07 by seandepagnier