pypilot/pypilot #58: turn on LCD screen and fix autopilot error compass failure

djdsails, 2020-05-11:
Hi,

Looking for help with the following issues:
1. turning on the lcd client, i have screen wired up per the pinout png on pypilot.org plug the 3v3 to VCC and ground to ground
2. need to know if IR or keypad need to be "turned on"? OR just plug them in and calibrate the IR
3. have compass failure - how can i fix this, steps please not a software engineer or coder. i have it wired up per the pinout png plus 3v3 to VCC and ground to ground
4. it also says no motor controller - can i wire up my own motor controller to pins 8 and 10, My own motor controller with a radio receiver like an RC car receiver,  will it work?

Thank you very much

seandepagnier, 2020-05-11:
2) on tinypilot ir is already working on pin 4
3) you need to wire i2c for compass sensors.
4) Yes

djdsails, 2020-05-11:
Hi sean,
still have a question on 1) how to turn on lcd
2) Great thanks
3) i have pin 3 on pi zero going to IMU SCL and pin 5 is going to IMU SDA
4) great thank you!

djdsails, 2020-05-11:
as you already know but link to the pinout i am looking at:

http://pypilot.org/wiki/lib/exe/detail.php?id=tinypilot&media=tinypilot_pinout.png

djdsails, 2020-05-11:
sean apologies number 1 now works, verified wiring, all good there screen works

djdsails, 2020-05-11:
i flipped the SCL and SDA pins to match this dwg: https://docs.sailoog.com/openplotter-v1-x-x/i2c/i2c-sensors and now it works...
hopefully my motor controller works, its a very simple controller, it just takes in input no feedback.
Does your Code for pypilot require feedback in order to clear the error??
Thank you
appreciate all the work you have done and will do

seandepagnier, 2020-05-11:
you can disable certain measurements with ifdefs to clear errors if
those analog pins are not set right.

freek4ever, 2020-05-13:
my screen briefly flashes text appears  wen i press the reset button  
as of now i have noting connected imu or motor controller  
is this normal i i have a imu wil the screen work 
i also have no conetion to opencpn runing on my pc   its conetted to te ricgt com port and when starting open cpn i see the arduino screen flash 

im geting a imu later this week hope this solves it 

love your post on hakaday

djdsails, 2020-05-19:
Hi Sean,
another question, Can pypilot work when it is in client mode and have it connect to the raspberry pi and work the same? i have wifi on the boat and cannot get the pi to work with two wifi sources. the pi recognizes the wifi dongle but no information is transferred.
any advice? 
i do have a serial USB serial on board, it might work but not sure if it will, i have no information on it baud rates etc

thank you!

djdsails, 2020-05-19:
IR sensor, pin 4 according to the pinout chart is a 5v out pin, why is the ir sensor configured to that 5v output? thanks

djdsails, 2020-05-19:
or do you mean GPIO #4?

djdsails, 2020-05-19:
also what is configure RF remote on the configuration tab on the internet browser, ie when logged into the pypilot ap... is RF remote frequency? could i use that instead of IR as its not working for me? any hardware additions needed? thank you!

djdsails, 2020-05-19:
wiring diagram for RF remote?

seandepagnier, 2020-05-19:
for rf remote you need to attach 433mhz receiver to arduino and
arduino to spi pins of raspberry.

closed 2021-04-06 by seandepagnier