pypilot/pypilot #59: MPU-925x Obsolescence

jdaandersj, 2020-05-11:
The MPU-9250 / MPU-9255 IMU series has been made obsolescent by the manufacturer for some time now.

Please could someone (@seandepagnier ?) give me an overview of the current barriers to switching to the ICM-20948? I would like to investigate what needs to be done to make it work and help if I can. 

TDK have an application note showing the differences: https://invensense.tdk.com/wp-content/uploads/2018/10/AN-000146-v2.0-TDK_Migration_MPU_9250toICM-20948.pdf

Many thanks for the great piece of software!

seandepagnier, 2020-05-11:
A "simple" (< 1000 lines) driver needs to be written.   This is a
little annoying because they appear to both use the same i2c
address...  So more detection logic is needed.

From what I can tell the only advantage of this chip is lower power
(which the different is irrelevant) and also 9 axis fusion which won't
be used anyway because the kalman filter runs on the pi.

The disadvantage is cost is $12 ??
I don't have this device yet, it would be nice to support but I do not
know if this is the best replacement for mpu9255.

Sean

jdaandersj, 2020-05-11:
Ok so detection of which IMU is used could be a problem - is there a particular reason not to make this user configurable (besides idiot-proofing)?

To me the main advantage is availability of new chips in the future, mpu9255 can still be bought easily in breakout form but not so easy to find from major component suppliers for use on custom pcbs.

I hadn't considered if the ICM-20948 would be the best replacement, perhaps this needs to be evaluated carefully. I naively assumed it would be the easiest to transition to since it seems to be TDK's intended successor for the mpu9255.

Let me know what you think, thanks for the fast response.

Jack

seandepagnier, 2020-05-11:
The software will have to probe the port to find if it is mpu9255 or
icm-20948.   So hopefully the user would not need to do anything.

For custom pcbs, it makes sense to use icm-20948 but it will cost more
than a breakout (you won't have to solder 4 pins)

jdaandersj, 2020-05-11:
Thanks for the response, I'd definitely like to help implement a new imu chip. I will have enough time to work on this fully from July, assuming you agree that it's a good idea (until then I have university work to do). If so, the imu chip/chips to add support for can be decided before then.

Please let me know if you would like me to have a go at this once I have time, and whenever you have time which imu/imus you think would be best to try (no rush).

seandepagnier, 2020-05-12:
I think it would be good to support this imu and really any others
that look interesting.  All we need to do is read the raw sensor
values out of i2c to support a new device.

Unless the sensors actually give better data than the mpu9255 I am not
concerned for 6-12 months from now because I will continue with
mpu9255.

sailoog, 2020-05-15:
I vote for LSM9DS1. It is already supported by RTIMULib2, it is in volume production by manufacturer and only few lines of code would be needed. Actually I have just done a successful calibration at office with latest RTIMULib2/pypilot. This is a gyros comparative LSM9DS1 vs MPU-9250 after calibration:

MPU-9250
[image]

LSM9DS1
[image]

There is really more noise in LSM9DS1 but the compass accuracy seems better.

Sean, is the compass accuracy evaluable by the scope?

seandepagnier, 2020-05-15:
Yes, you can evaluate compass but take into consideration the total
range for the sensors compared.

This is exactly what I remembered.  The mpu9255 has better gyro but
worse compass.   The compass sensors in the mpu9255 are not very good.
  The gyro sensors are a lot more important than compass.

romton843, 2020-09-15:
Hi everyone,
Regarding the gyro sensors, here is a comparison of the datasheets of some of the ones mentioned above:
[](https://learn.adafruit.com/comparing-gyroscope-datasheets)
The FXAS21002C seems to be a better option but I don't think it is supported by RTIMULib2.
However, the BNO055 (with the BMI055 gyro) is actually supported by RTIMULib2, isn't it?
Any thoughts regarding this last one?
Best regards and thank you all for this awesome project.
Romain

seandepagnier, 2020-09-16:
I have ordered a few possible replacements which are reportedly much better sensors than the mpu9255, but they also cost $15-$20   I think the upgrade may be worth it for the improved gyros.

Where do you get the FXAS21002C?   The BNO055 is apparently worse than the mpu9255 so I am not very curious about it, although I do have one I never got around to trying it.

romton843, 2020-09-16:
Oh, I just looked up on NPX semiconductors website (the manufacturer) and it looks like it's also been discontinued... Sorry.
I have one that I'm not using though, this one: https://www.adafruit.com/product/3463?gclid=CjwKCAjwzIH7BRAbEiwAoDxxTu1xhEcM32gecMg33jFtwknpt9cZb_AojSRD-49Zl6cVK2QAohQ-mhoC1rIQAvD_BwE
Would you be interested in me sending it to you?

seandepagnier, 2020-09-16:
This looks interesting too, you could send it to me but I don't know
when I will get to try it I have ordered
1 x Adafruit ISM330DHCX + LIS3MDL
as well as
 1 x Adafruit TDK InvenSense ICM-20948 9-DoF IMU (MPU-9250 Upgrade) (STEMMA

the more sensors supported the better.

This sensor seems to have the lowest noise levels but I have not had
the chance to implement it in rtimulib2 yet.

Sean

romton843, 2020-09-16:
I'd be happy to contribute. What address should I use for mailing it?

seandepagnier, 2020-09-16:
send me email seandepagnier at gmail dot com

orca-hydromancer, 2020-11-18:
Hi.

I have prelim version of ICM-20948 imu driver at https://github.com/orca-io/RTIMULib2/commit/80f17fb80f331a3c6a7f320919705da3547ea275

I've successfully used it with pypilot. By that I mean it outputs reasonable values. Still need to check if the fusion is as smooth and appropriate (for boat motion) as was with MPU9250.

However, there are few considerations:

1. I'm inexperienced in writing drivers and reading datasheets so this was a learning experience
2. My process was to map MPU9250 driver 1:1 by comparing datasheet of MPU9250 and ICM20948
3. The FIFO mode I couldn't get working. I just couldn't map the behavior of MPU9250 FIFO as not all registers are the same. I'd really appreciate someone capable having a fresh look at it and giving it a try.
4. ICM20948 has multiple register banks, and issuing ops to registers requires selecting proper bank that register belongs to before doing the op (unless you are certain that proper bank is selected already). WHO_AM_I register is available only in BANK 0, and BANK 0 is default bank only on powerup of the device, changes during runtime based on ops done, so to make sure that device can be detected properly on RTIMULib startup, BANK 0 needs to be selected before even checking device identity. So, to me this seems a bit absurd: to detect device I need to reset selected bank, but to do that I need to know that it is ICM20948 device, so not to corrupt some other sensor device. Currently I reset selected bank externally before running my program with `i2cset -y 1 0x68 0x7F 0x00 && ./program_using_rtimulib2`, but I know for sure that is the device I have. If anyone has any ideas on how to do this better, please come forward.

If anyone is interested in looking at FIFO, more testing of the driver, boatimu.py integration, and making this proper, I'm happy to collaborate.

The references I've used:
https://invensense.tdk.com/wp-content/uploads/2016/06/DS-000189-ICM-20948-v1.3.pdf
https://invensense.tdk.com/wp-content/uploads/2015/12/eMD_Software_Guide_Nucleo_ICM20x48.pdf
https://github.com/pimoroni/icm20948-python/blob/master/library/icm20948/__init__.py
https://github.com/PX4/Firmware/blob/master/src/drivers/imu/invensense/icm20948/InvenSense_ICM20948_registers.hpp

jdaandersj, 2020-11-18:
Good work, I will look through it properly and see if I can add anything as soon as I have the time. I'm sorry I haven't contributed anything to my own issue yet! Covid has completely screwed up what plans I had for working on this.

sailoog, 2020-11-21:
Good work! not yet tested, waiting for an ICM-20948 module. We really need this to move forward with the integration of pypilot into OpenPlotter.

orca-hydromancer, 2020-12-01:
Couple of questions:

1. Do I understand this correctly:
- RTIMULib2 expects gyro and magnetometer axes as: x fwd, y right, z down; accel as: x back, y left, z up
- I got this understanding by comparing BNO055 and MPU-9250 drivers and orientations defined in respective datasheets
- Magnetometers' orientation of axes is diff for MPU-9250 and ICM-20948. ICM-20948 magnetometer is oriented already as the lib expects so there is no need to swap X and Y as is done for MPU-9250.

Orientation of Axes for ICM-20948:
[image]

Orientation of Axes for MPU-9250:
[image]

2. Is there any impact of the above to pypilot's boatimu? I can calibrate magnetometer with RTIMULibDemoGL using min/max and get good heading readings. However, can't get satble heading by using boatimu and pypilot_calibration. Are there any MPU-9250 in pypilot specifics to consider?

seandepagnier, 2020-12-01:
there really should not be any mpu9250 specifics in pypilot.   So i'm
not sure what is going wrong.

Using min/max calibration normally does not give quite as good results
and is much more prone to errors.   It can't be done automatically.
So I would really compare the bias and scale factors that the
calibration computes.

What do you mean not a stable heading?   Maybe the driver isn't
working correctly?   Some output logs would be helpful.

I also think it's really important to use the fifo mode of the imu, or
else it won't get nearly as precise of measurements because it cannot
average.   With the 9250 or 9255, it takes 100 samples per second, but
averages them into what is read by pypilot which greatly reduces
noise.

orca-hydromancer, 2020-12-28:
> What do you mean not a stable heading?   Maybe the driver isn't
working correctly?   Some output logs would be helpful.
I meant that the heading reading was unpredictable with lot of drift which is expected in uncalibrated mode. I couldn't get to calibrated mode (even after 5 minutes of trying) with pypilot's ellipsoid fit calibration, will provide some screenshots later. My base test is RTIMULibDemoGL using min/max calibration which produces correct heading readings and the fusion works as expected. Currently I combine RTIMULib's min/max with pypilot's level alignment functionality. Not ideal for reasons you mentioned.

> I also think it's really important to use the fifo mode of the imu, or
else it won't get nearly as precise of measurements because it cannot
average
I agree FIFO mode is crucial, however I'm in over my head at that point. Currently my driver is suffering from performance, using too much CPU between reading the two chips. So, really this could use some fresh capable mind(s).

seandepagnier, 2020-12-28:
I have got it working, but somehow a small bug exists that occurs only
once every 14 hours and I have to reset the device, so it's taking me
a while to debug.

With fifo mode, the sensors prove to be identical in noise level to
the mpu9255.   The calibration is also working.

orca-hydromancer, 2020-12-28:
Wow, great news.

I'm happy to try it out and help debug if possible.

gitter271, 2021-01-24:
Hi, just wanted to share that I got this CJMCU-9680 board working on the raspberry pi with openplotter: 
https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20210124141854&origin=y&SearchText=CJMCU-9680

It has the BME680 sensor which works in Signal K using a plugin and the ICM-20948 sensor, which I got working in Python using the adafruit drivers. Any progress on getting it to work in PyPilot/Openplotter?
I was also wondering if the ICM-20948 could be part of the I2C application in Openplotter. Perhaps PyPilot could read the data from there using signal K.
Let me know if I can send this board to some developers, I'm happy to help out.

romton843, 2021-01-24:
Hi, regarding the BME680, I just updated the Signal K plugin called Raspberry-Pi BME680 which brings pressure, temperature, humidity and air quality to Signal K and therefore to your NMEA network. It’s a great addition to your system since it gives you a way to get notified if there is a gas leak or for carbon monoxyde.
I don’t know if it’s updated on Signal K yet but you can find it here:
https://github.com/oehoe/signalk-raspberry-pi-bme680

gitter271, 2021-01-24:
That's indeed the plugin I used. It's part of the Signal K plugin collection. Thanks, works great!

seandepagnier, 2021-01-24:
>  Any progress on getting it to work in PyPilot/Openplotter?

It works now on my RTIMULib2 driver however I plan to finalize the implementation to make it more robust.

> I was also wondering if the ICM-20948 could be part of the I2C application in Openplotter. Perhaps PyPilot could read the data from there using signal K.

openplotter can read the bme680 directly since this data is not used by pypilot anyway.    It is not good for pypilot to get the imu data from signalk because of too much latency.    It is absolutely critical that pypilot reads from the gyros in the realtime process and the latency is guaranteed lowest possible.

Having the two share the same i2c bus is probably ok although I would be careful what uses this bus that it does not access it for too long (thus reducing latency again)

sailoog, 2021-01-25:
> It has the BME680 sensor which works in Signal K using a plugin and the ICM-20948 sensor, which I got working in Python using the adafruit drivers. Any progress on getting it to work in PyPilot/Openplotter?

adafruit drivers just do the easiest part of getting raw data from the IMU. You also need the complex part, processing raw data and calibration.

> I was also wondering if the ICM-20948 could be part of the I2C application in Openplotter. Perhaps PyPilot could read the data from there using signal K.

No matters if we add IMU management to openplotter-i2c app or openplotter-pypilot, we will always need pypilot to process IMU data, so it makes no sense to add IMU management to both apps.

On the other hand pypilot is only optimized for the old mpu9255/9250 and soon for the new icm-20948. The rest of IMUs supported by RTIMULib2 driver will work but they will not use pypilot optimization. RTIMULib2 driver has its own calibration and data processing tools but it is not as accurate as pypilot is and it has not good auto-calibration. Maybe adding the RTIMULib2 way to openplotter-i2c could be an option for people using not supported IMUs who do not need accuracy (no autopilot and just aprox magnetic heading data).

seandepagnier, 2021-01-26:
It works for the icm20948 in my rtimulib2, however occasionally (after
hours or infrequently) the fifo gets out of sync.   This was an issue
with the 9255 and I detected it by out of range compass data but for
the icm20948 I want to do even better and query who am i byte from the
compass each sample (only 1 byte) to verify data alignment, so I
didn't do this yet.

The icm20948 appears to have identical quality of raw values from
sensors of the 9255 so not much reason to upgrade unless you cannot
obtain the 9255.

I added support for ISM330DHCX + LIS3MDL as well which has improved
gyro and accelerometers but the compass is not better.    I think this
board is maybe worth it, but I would try to find a better compass
sensor than the LIS3MDL.   I have not fully optimized this driver, so
I would expect the data to improve (at least for compass) if I did.

I also realize other drivers in rtimulib2 can be optimized.   This
mostly means to sample them faster into their internal fifo, then read
all of the samples and average them at the normal rate which gives
lowest cpu and noise level.

The key to pypilot for imu processing is contained in
calibration_fit.py   It uses regression to automatically fit the
compass, and I use a special formula which helps give more accurate
bias to compass by utilizing the accelerometers.

seandepagnier, 2021-08-07:
The replacement for mpu9255 is icm20948.   This part is no better and requires 1.8v levels but at least the quality seems better without so many with failed gyros.   Unfortunately the price is doubled since 2021

closed 2021-08-07 by seandepagnier