pypilot/pypilot #61: Debian packaging.

FredericGuilbault, 2020-08-31:
Hi, I was about to create a Debian package of pypilot to accelerate my OS build time. Then I saw that @sailoog  was also doing the same but with some openplotter's specific config in it.

So in order to avoid work duplication, If there was a implementation neutral Debian package that would exist. Anyone interested could use it as-is or fork the code or write a wrapping meta-package. But share a common base.   

I understand that you use Tiny core and are probably not so motivated about debian packages. But im about to do it anyway for lysmarine. So if you would agree be the owner (and subsequently the benevolent dictator) of the feature/repository. I would prefer to do a Pull request to pypilot and make everyone profit from it then having it for the sole purpose of my project. 

--Fred

seandepagnier, 2020-09-25:
This would be great.   There are already some scripts for debian as pypilot provides a few system services.   You may need to include the pypilot_data repository in the package script or make it a separate package (upgrading pypilot won't need to update this)

sailoog, 2020-09-27:
Go ahead if you want, finally we are compiling in openplotter instead of using a debian package because pypilot is too often updated but a maintained package would be great.
Some advices:
- Use python3
- Use Debian packages for dependencies when possible instead of pip3 packages.
- There are some dependencies not present in README like this one: https://github.com/pypilot/pypilot/blob/18396322f8c47c5d78417b9bd904713b634c858d/setup.py#L31

FredericGuilbault, 2020-09-27:
@seandepagnier It would mean adding a `DEBIAN` folder to your repository. Another way to do it is to create a separate repository to with all the debian stuff and a script that pull from pypilot's repos. 

The first option make it more concise and KISS, The second keep concerns separated.

Official Debian packaging doc prefer the second one as they are implying that packagers are dealing with someone else code without access to the genuine repository(they call it upstream repo). But their toolset and documentation support both ways. 

What do you prefer ?

FredericGuilbault, 2020-09-27:
> pypilot is too often updated but a maintained package would be great. 
At one point, and the packaging will be pushing in that way. Pypilot will need a dev branch and release versioning.

rgleason, 2020-09-27:
Why not use the OpenCPN Plugin Manager system which is working now, with Pypilot in the master catalog?
The cloudsmith repository for pypilot is here
https://cloudsmith.io/~opencpn/repos/pypilot-beta/packages/?q=0.22.5.0*tarball
You can load the tarballs using Options>Plugin>  "Import Plugin"

What build do you need that isn't here already?
Just do a search on opencpn packages and pick the Deb packages for raspbian!
https://cloudsmith.io/~opencpn/repos/pypilot-beta/packages/?q=opencpn*0.22.5.0

seandepagnier, 2020-09-28:
@rgleason 
I believe he is interested in a debian package of pypilot itself, not the opencpn plugin

@FredericGuilbault
Could you just use a separate repository for now?   Otherwise, how much data is in the DEBIAN folder?   If it's < 100kb I don't have much concern as long as it won't grow either, but otherwise I do not wish to increase clone times significantly, this is why the pypilot_data repository exists.

FredericGuilbault, 2020-09-28:
The last one I made was 30Kb for rtl-ais. ...It's just a bunch of config files really. 

But yes, I would also vote for a separate repository.  Can you create it. Ill fork it, add the stuff and PR 

---
@rgleason 
 From my sysadmin point of vue. In a perfect world, if Opencpn want to interact with pypilot. it should be done thru systemd/v and a API/IPC. Not in a parent child relationship.

pypilot is small compared to Opencpn, Launching Opencpn just to run pypilot is like a 9000% overhead. [Insert it's over 9000 meme here] 

This said, ill take a deep look at the debian package you have on cloudsmith. Thx for sharing.

rgleason, 2020-09-28:
@FredericGuilbault  Yes, I understand your point about OpenCPN overhead just to run pypilot. Some Windows guys simply don't care, they must have unlimited power.

Off-topic: Do you have rtl-ais working for rpi?  I have never been able to get this going in Windows and would very much like too.  -Do you have any suggestions for me?   ..Also that rtlsdr plugin of Sean's is probably one of the next ones to redo for Plugin Manager.

sailoog, 2020-09-29:
@FredericGuilbault do not forget that you will need to maintain a second package for RTIMULib2 as dependency of the pypilot package. Raspbian already contains a rtimulib package that must be uninstalled before installing/compiling RTIMULib2.
Question: how can debian packages deal with pip3 dependencies?

seandepagnier, 2020-10-04:
@FredericGuilbault I created https://github.com/seandepagnier/pypilot-debian

closed 2020-10-04 by seandepagnier

FredericGuilbault, 2020-10-06:
@seandepagnier "Question: how can debian packages deal with pip3 dependencies?" 

I have no idea yet. I might have to go annoy other python project as well :)