pypilot/pypilot #63: No module named 'pilots'   No module named 'pyjson'

FredericGuilbault, 2020-10-07:
Hi, when Im running unit testing with the following command:

`python3.8 -m unittest discover -v`

And it says:
```
ERROR loading absolute.py No module named 'pilots'   No module named 'pyjson'
ERROR loading simple.py No module named 'pilots'   No module named 'pyjson'
ERROR loading basic.py No module named 'pilots'   No module named 'pyjson'
ERROR loading learning.py No module named 'pilots'   No module named 'pyjson'
```
Agan, im no python dev. But correcting the line 11 to   `import pypilot.pyjson` in `pypilot/values.py`

and correcting the line 11 to   `from pypilot.resolv import resolv` in `pypilot/pilots/basic.py`

Suppress the error message. But idk what else it could impact. Im still confused about relative and full path to import files in python.

FredericGuilbault, 2020-11-12:
I ran the test agan today and a 5th error now show 
```
ERROR loading intellect.py No module named 'pilots'   invalid syntax (intellect.py, line 205)
```

seandepagnier, 2021-03-05:
You ran "python3.8 -m unittest discover -v" in the pypilot directory and expected it to do something useful?

FredericGuilbault, 2021-03-06:
Im not sure to understand the question.   I mean, I expect unit test to success. 

As part of my effort to make pypilot Debian package, the packaging tools provided by the Debian team run the unit tests. And stop if test fails. I digged it and they use pretty much `python3.8 -m unittest discover -v`

seandepagnier, 2021-04-06:
well, there are no unittests for pypilot currently, so I don't really expect this to work

closed 2021-04-06 by seandepagnier