pypilot/pypilot #67: Git install on openplotter

Jamos1988, 2020-11-10:
I've uninstalled Pypilot 0.16 (newest one available on Openplotter app) and installed the newest 0.23 through git.
Installation went fine.

Pypilot service started only after rebooting the pi.
But, the pypilot_webapp still does not start:

` pypilot_webapp.service - pypilot web
   Loaded: loaded (/etc/systemd/system/pypilot_webapp.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-11-10 12:37:26 CET; 2s ago
  Process: 4964 ExecStart=/usr/bin/sudo pypilot_web (code=exited, status=1/FAILURE)
 Main PID: 4964 (code=exited, status=1/FAILURE)

Nov 10 12:37:26 openplotter systemd[1]: pypilot_webapp.service: Main process exited, code=exited, status=1/FAILURE
Nov 10 12:37:26 openplotter systemd[1]: pypilot_webapp.service: Failed with result 'exit-code'.
`

Not sure what this means.

FredericGuilbault, 2020-11-29:
It mean the service fail to start pypilot_webapp   by using the following launch command : `/usr/bin/sudo pypilot_web`

Have you try to launch it manually form the terminal like this `sudo pypilot_web` It will probably create a more meaningful error message.

marcobergman, 2021-04-09:
Probably related to #79; after that fix, it works now
Issue turned stale, suggest closing this issue

`pi@openplotter:~/pypilot/pypilot $ sudo systemctl start pypilot_web
pi@openplotter:~/pypilot/pypilot $ sudo systemctl status pypilot_web
● pypilot_web.service - pypilot_web
   Loaded: loaded (/etc/systemd/system/pypilot_web.service; disabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-04-09 20:59:14 BST; 10s ago
 Main PID: 31697 (pypilot_web)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/pypilot_web.service
           └─31697 /usr/bin/python3 /usr/local/bin/pypilot_web 8080

Apr 09 20:59:14 openplotter systemd[1]: Started pypilot_web.
`

closed 2021-08-07 by seandepagnier