pypilot/pypilot #69: Log growth on error conditions is an issue

mgrouch, 2020-12-02:
PyPilot repeatedly complains about I2C availability and error logs grow fast very soon.
PyPilot should not report recurring error conditions in the log so often otherwise
it would create another error condition with the amount of free space.

See also 
https://github.com/lysmarine/lysmarine_gen/issues/97

seandepagnier, 2020-12-02:
which error is emitted?   Do you have an example of the log?

I think it is actually rtimulib2 complaining since pypilot does not directly access i2c.   It will continue to try to find sensors again and again, so if you plug one it will discover it.   I will be able to modify rtimulib2 to not repeat the error message.

FredericGuilbault, 2020-12-09:
I think this is the logs in question.  
But on my case, I don't have anything plugged in the RPI...  

```
Dec  9 16:50:53 lysmarine pypilot[416]: Using fusion algorithm Kalman STATE4
Dec  9 16:50:54 lysmarine pypilot[416]: Failed to open I2C bus 1
Dec  9 16:50:54 lysmarine pypilot[416]: Failed to open SPI bus 0, select 0
Dec  9 16:50:54 lysmarine pypilot[416]: Failed to open SPI bus 0, select 1
Dec  9 16:50:54 lysmarine pypilot[416]: No IMU detected

Dec  9 16:50:54 lysmarine pypilot[416]: Using fusion algorithm Kalman STATE4
Dec  9 16:50:55 lysmarine pypilot[416]: Failed to open I2C bus 1
Dec  9 16:50:55 lysmarine pypilot[416]: Failed to open SPI bus 0, select 0
Dec  9 16:50:55 lysmarine pypilot[416]: Failed to open SPI bus 0, select 1
Dec  9 16:50:55 lysmarine pypilot[416]: No IMU detected

Dec  9 16:50:55 lysmarine pypilot[416]: Using fusion algorithm Kalman STATE4
Dec  9 16:50:56 lysmarine pypilot[416]: Failed to open I2C bus 1
Dec  9 16:50:56 lysmarine pypilot[416]: Failed to open SPI bus 0, select 0
Dec  9 16:50:56 lysmarine pypilot[416]: Failed to open SPI bus 0, select 1
Dec  9 16:50:56 lysmarine pypilot[416]: No IMU detected
```

seandepagnier, 2021-04-10:
This should be solved with my changes to pypilot and rtimulib2 now

closed 2021-04-10 by seandepagnier