pypilot/pypilot #70: signalk synchronisation fails over track message, expecting xte

marcobergman, 2020-12-30:
After the patch of 30-dec-2020, pypilot seems to be resistant against these 1.36 meta messages (ref https://forum.openmarine.net/showthread.php?tid=3086&pid=17534#pid17534)

Now, there’s a different error:

2020-03-14_11:01:58.40393 Traceback (most recent call last):
2020-03-14_11:01:58.40403   File "/tmp/tcloop/pypilot/usr/local/lib/python3.6/site-packages/pypilot/signalk.py", line 242, in poll
2020-03-14_11:01:58.40417     self.sensors.write(sensor, data, 'signalk')
2020-03-14_11:01:58.40425   File "/tmp/tcloop/pypilot/usr/local/lib/python3.6/site-packages/pypilot/sensors.py", line 187, in write
2020-03-14_11:01:58.40442     self.sensors[sensor].write(data, source)
2020-03-14_11:01:58.40451   File "/tmp/tcloop/pypilot/usr/local/lib/python3.6/site-packages/pypilot/sensors.py", line 38, in write
2020-03-14_11:01:58.40465     self.update(data)
2020-03-14_11:01:58.40469   File "/tmp/tcloop/pypilot/usr/local/lib/python3.6/site-packages/pypilot/sensors.py", line 95, in update
2020-03-14_11:01:58.40483     self.xte.update(data['xte'])
2020-03-14_11:01:58.40492 KeyError: 'xte'

this was the offending message:
{'track': 144.99200003310594, 'device': 'EC'}

marcobergman, 2021-01-06:
Please note that **this resets and disconnects pypilot repeatedly**, quite an unwanted effect of the signalk integration. Due to #74, this condition is only resolved until signalk is restarted.

seandepagnier, 2021-01-06:
I pushed changes to address this, is it still occurring?

marcobergman, 2021-01-07:
Ok saw the change, rolled it out, and the issue no longer occurs. Pypilot no longer resets, and seems to get both track and xte data from opencpn though signalk. 

I do have these messages in my log, still:
2020-03-14_11:27:31.33915 Exception converting signalk->pypilot unsupported operand type(s) for /: 'dict' and 'float' {'unknown': {'steering.autopilot.target.headingTrue': {'units': 'rad', 'description': 'Target heading for autopilot, relative to North'}}, 'II': {}, 'pypilot': {}, 'EC': {}, 'GP': {}}

seandepagnier, 2021-01-07:
Good, so it prints errors instead of throwing exceptions..

It seems like signalk is a moving target at this point, not sure what
I should do about it, maybe just use nmea0183 for autopilot route
following in the meantime.

Sean

closed 2021-04-06 by seandepagnier