pypilot/pypilot #71: motor.ino wrong measurements when pwm_style == 2

marcobergman, 2020-12-30:
If I define VNH2SP30 in the latest motor.ino from github, the measured controller voltage goes down by a factor 4.7 (resulting in BAD VOLTAGE), but the controller temp goes up by a factor 2.6. If I undefine VNH2SP30, I get correct readings for both measures. To use the pwm_style=2 modus without messing up or disabling the readings, it seems to suffice to drop 'pwm_style == 2 || ' in line 455 of the current version

`    // setup adc
    DIDR0 = 0x3f; // disable all digital io on analog pins
    if(pwm_style == 2 || ratiometric_mode)
        adcref = _BV(REFS0); // 5v
    else
        adcref = _BV(REFS0)| _BV(REFS1); // 1.1v`

marcobergman, 2020-12-30:
Ref: https://forum.openmarine.net/showthread.php?tid=1561&pid=13182#pid13182

seandepagnier, 2020-12-31:
well it's difficult to support all possible combinations since you
likely have different resistances for your sensors.

It may be better you make a new #define for your intended setup and
then it can be supported for whatever you like to use.

CapnKernel, 2021-01-02:
This isn't about different setups, this is a BUG: Readings are just _wrong_ for the VNH2SP30 and VNH5019.  100% of people trying the VNH2SP30 and VNH5019 will have this problem.

I talked about it here and analysed the circuit and code:

 https://forum.openmarine.net/showthread.php?tid=2868&pid=16359#pid16359

seandepagnier, 2021-01-03:
where is the bug?   Sorry what exactly should change?    Depending on
the configuration it's assumed you use a resistor divider on the
current sense.

CapnKernel, 2021-01-03:
> Depending on the configuration it's assumed you use a resistor divider on the current sense.

Yes, you need a resistor divider, and the motor controller schematic gives values of 47k/120k.  The problem is that anyone who uses a VNH2SP30 or VNH5019 with those values _and_ the stock code will have the problem.  100% fail rate.  I found it, Timo found it, marcobergman found it, and there have been other reports too in the forum.  Perhaps you didn't read my analysis:

  https://forum.openmarine.net/showthread.php?tid=2868&pid=16359#pid16359

> where is the bug?  

The bug is that if someone uses the stock code, and the resistor values from the motor controller schematic, the readings will be wrong, as I wrote about here:

> Sorry what exactly should change?

Please accept marcobergman's changes :-)

seandepagnier, 2021-01-04:
the motor controller schematic doesn't use the vnh2sp30 which is
probably why I'm confused.

In any case, I am more than willing to accept pull requests that only
affect pwm_style==2 and don't change the style 0 or 1 operation in any
way

marcobergman, 2021-02-01:
Fixed in 03-jan-2021 commit; issue closed. Still some problem with current measuring but I'll make another issue.

closed 2021-02-01 by marcobergman