pypilot/pypilot #74: signalk synchronization uses stale data

marcobergman, 2021-01-06:
The signalk synchronization, set up through the zeroconf facility, does not seem to take the timestamp into consideration. When you switch **off** an active route in opencpn, the data that was derived by signalk from the APB message becomes stale, and the timestamp on those data elements in signalk appears frozen. Nevertheless, the data keeps on being synchronized into pypilot. At some point, this synchonization into pypilot should stop, because the data is no longer valid. This becomes visible because of issue #70 that continues crashing pypilot even though there are no active route any more. Restarting signalk is the only option, but this is far from apparent for the ordinary user.

seandepagnier, 2021-04-06:
So, opencpn sends apb track message to signalk which converts it into a signalk message.

What I dont understand is why would the sigalk server continue to re-transmit stale messages?   Isn't the bug in signalk-node or the nmea to signalk conversion part which is outside of pypilot?

seandepagnier, 2021-08-07:
Any update on this issue?   Were you following routes using signalk??    I would like to get this working and switch over to signalk control to eliminate the need for nmea data.

rgleason, 2025-02-11:
Marco can this be closed? Or is it still an issue?

marcobergman, 2025-02-11:
Can't replicate it anymore, so closed it. Pypilot seems to _set_ steering.autopilot.target.headingTrue now, when its receives APB messages though 20220. When I was investigating this, I was trying pypilot to _read_ this key from signalk. I have stopped trying doing NAV mode through signalk now, and for all I know I was the only one who tried to do so.