pypilot/pypilot #75: Exception in Hat with custom_0 in Tinypilot

thomasnisole, 2021-01-06:
Hello Sean, 

I installed tinypilot on Rpi 0 and i searched to wire à tft screen (i saw a tft condition in code) but when i changed hat conf file to take tft driver, it has been reset because an exception is throw by the line :

https://github.com/pypilot/pypilot/blob/784eb01c0c65be5a31fb3eeed145262910be1001/hat/hat.py#L201

Have you an idea ? 

Thx.

seandepagnier, 2021-01-07:
This file only exists if there is an i2c hat eeprom with a config file which describes the lcd

so if you don't have this file, it should fail to load it but not reset the program.   Instead it will look in ~/.pypilot/hat.conf

The "tft" driver is for the esp32 running micropython.   It is not supported in linux.   Instead, maybe you can use "framebuffer" if the tft screen you have is supported by a linux framebuffer driver and you have got that working.

thomasnisole, 2021-01-07:
Thx @seandepagnier,

I has already test with framebuffer (with hdmi screen), but my configuration was always override at restart

seandepagnier, 2021-01-07:
did it work with framebuffer for the tft screen?

Did you edit ~/.pypilot/hat.conf

Sean

thomasnisole, 2021-01-11:
No the tft screen is not compatible. I test with an hdmi screen, but when i edit the hat conf file is always override on restart. 
I continue my test.

Thx

seandepagnier, 2021-01-12:
I will try to get hdmi screen working..  It was the original screen
before I started using spi display.  It should not be overwritten.

As for the screen, are you sure you can't get a linux framebuffer
driver like "fbtft" support?

closed 2021-08-07 by seandepagnier