pypilot/pypilot #79: pypilot_web on openplotter fails, 'unsupported version of the Socket.IO or Engine.IO'

marcobergman, 2021-03-13:
Symptom: pypilot Server: Not Connected.

pypilot_web output:
```
pi@openplotter:~ $ pypilot_web
using default port of 8000
processing clients
The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)
```
Javascript console:
[image]

marcobergman, 2021-03-13:
pypilot_web works on tinypilot, but not on openplotter. That makes sense; the tinypilot libraries are different.

https://github.com/miguelgrinberg/python-socketio/issues/578

pi@openplotter:~/pypilot/web/static $ pip freeze
Flask-SocketIO==2.9.0
python-socketio==5.0.4
python-engineio==4.0.0

JavaScript Socket.IO version | Socket.IO protocol revision | Engine.IO protocol revision | Flask-SocketIO version | python-socketio version | python-engineio version
-- | -- | -- | -- | -- | --
0.9.x | 1, 2 | 1, 2 | Not supported | Not supported | Not supported
1.x and 2.x | 3, 4 | 3 | 4.x | 4.x | 3.x
3.x | 5 | 4 | 5.x | 5.x | 4.x

marcobergman, 2021-03-13:
I've tried to upgrade all these packages but to no avail:
```
pip3 install -U Flask-SocketIO python-socketio python-engineio
pip3 freeze|grep -iE "engine|socket"
Flask-SocketIO==5.0.1
python-engineio==4.0.1
python-socketio==5.1.0
```

marcobergman, 2021-03-13:
It looks like this does the trick:
```
pip uninstall python-socketio python-engineio flask-socketio
pip install "python-socketio<5"
```
Resulting in 
```
pi@openplotter:~/python-socketio $ pypilot_web
using default port of 8000
processing clients
Client connected fc9c66251aa942ffbe40a91b1a5509e8
```

seandepagnier, 2021-03-13:
Thanks for finding this!

Would another option be to replace the javascript socketio library
with a compatible version?

marcobergman, 2021-03-13:
May be; this was simply the first thing that worked. I could only find these three python libraries under pip; I would not know where the javascript library sits. If I do pip install "python-socketio" (that is, without the <5) I have the same problem.

sasadrnjevic, 2021-03-21:
I have same  symptom, but trick didn't work

pi@raspberrypi:~ $ pip uninstall python-socketio python-engineio
Skipping python-engineio as it is not installed.
Skipping python-socketio as it is not installed.
pi@raspberrypi:~ $ pip install "python-socketio<5"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python-socketio<5
  Downloading https://files.pythonhosted.org/packages/b2/91/d78afbe5861de53cafe9dcc6eb1a802d5495cfbffb79b8385ee7cf15a71c/python_socketio-4.6.1-py2.py3-none-any.whl (51kB)
    100% |████████████████████████████████| 61kB 424kB/s 
Requirement already satisfied: six>=1.9.0 in /usr/lib/python2.7/dist-packages (from python-socketio<5) (1.12.0)
Collecting python-engineio<4,>=3.13.0 (from python-socketio<5)
  Downloading https://files.pythonhosted.org/packages/8d/35/b9edf38b6a42077f3c097ab4edb0281adb3cc5434645c0bd301b0b45c767/python_engineio-3.14.2-py2.py3-none-any.whl (51kB)
    100% |████████████████████████████████| 61kB 945kB/s 
Installing collected packages: python-engineio, python-socketio
Successfully installed python-engineio-3.14.2 python-socketio-4.6.1
pi@raspberrypi:~ $ pip freeze|grep -iE "engine|socket"
python-engineio==3.14.2
python-socketio==4.6.1
pi@raspberrypi:~ $ pypilot_web
using default port of 8000
processing clients
The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)

any suggestion...

sasadrnjevic, 2021-03-21:
success...

pi@raspberrypi:~ $ pip3 freeze | grep -iE "engine|socket"
Flask-SocketIO==5.0.1
gevent-websocket==0.10.1
python-engineio==3.14.2
python-socketio==4.6.1
websocket-client==0.58.0
pi@raspberrypi:~ $ pip3 uninstall Flask-SocketIO
Uninstalling Flask-SocketIO-5.0.1:
  Would remove:
    /home/pi/.local/lib/python3.7/site-packages/Flask_SocketIO-5.0.1.dist-info/*
    /home/pi/.local/lib/python3.7/site-packages/flask_socketio/*
Proceed (y/n)? y
  Successfully uninstalled Flask-SocketIO-5.0.1
pi@raspberrypi:~ $ pip3 freeze | grep -iE "engine|socket"
Flask-SocketIO==2.9.0
gevent-websocket==0.10.1
python-engineio==3.14.2
python-socketio==4.6.1
websocket-client==0.58.0
pi@raspberrypi:~ $ pypilot_web
using default port of 8000
processing clients
Client connected 74ae2fb66d3240cdb525db829d5fe9dc
connect pypilot...
connected to pypilot server
post request handler error
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/engineio/server.py", line 418, in handle_request
    socket.handle_post_request(environ)
  File "/home/pi/.local/lib/python3.7/site-packages/engineio/socket.py", line 128, in handle_post_request
    p = payload.Payload(encoded_payload=body)
  File "/home/pi/.local/lib/python3.7/site-packages/engineio/payload.py", line 15, in __init__
    self.decode(encoded_payload)
  File "/home/pi/.local/lib/python3.7/site-packages/engineio/payload.py", line 76, in decode
    raise ValueError('Too many packets in payload')
ValueError: Too many packets in payload
closed pypilot client
Client disconnected 74ae2fb66d3240cdb525db829d5fe9dc 0
Client connected 1e920441076e4bf6a98dbd9f6cf6df1d
connect pypilot...
connected to pypilot server

Thank You!

marcobergman, 2021-03-21:
maybe it's a python2 / python3 thing. It is weird that is was not installed in the first place. Try pip3 instead of pip?

marcobergman, 2021-03-21:
You beat me to it ;-)

seandepagnier, 2021-03-21:
Now you understand the problem and I don't.    Is this issue likely to
continue in the future, is there something that needs to change in
pypilot?

marcobergman, 2021-03-22:
Oh I have no classical understanding of this all; I just try 10 things and then one works. My assumption here was that python2 and python3 have different library paths, so when you need a library in python3, installing it with a python2 installer won't work.
For future, I think to rely on explicit naming of the python executable (using python3 instead of python) is advisable; not to be dependent on what the combined outcome of python, debian and image (openplotter) installers might be in the case of parallel 2/3 installs, which seems to be the standard for the next years, until python2 is really out of the picture.

marcobergman, 2021-04-09:
To come back to your question, pip normally knows what libraries are compatible with each other, but it looks like in this case it had to be helped a bit by limiting the version of python-socketio to stay below 5. I don't know how the openplotter dependencies mechanism works, but you would need to keep an eye onto this, and test it. I cannot see python-socketio at the moment in dependencies.py, and I don't know how this script interacts with openplotter. I have a small ubuntu machine here that is rather pristine; I'll run your updated setup.py on it and see how it fares in this matter.

marcobergman, 2021-04-10:
Just installed on a pristine ubuntu server - it seems dependencies.py is lacking flask_socketio. I can't get it installed here, see attached. Furthermore, the pypilot logging mentions `failed import calibration fit No module named 'scipy'` so you might want to check that dependency as well.

[pypilot_web_ubuntu.txt](https://github.com/pypilot/pypilot/files/6290291/pypilot_web_ubuntu.txt)

seandepagnier, 2021-04-10:
I tried on a similar machine and it works.   Please review this log and try it on your own ubuntu.
[pypilot_web.txt](https://github.com/pypilot/pypilot/files/6290396/pypilot_web.txt)

marcobergman, 2021-04-10:
So it boils down to simply installing **flask**-socketio<5 instead of **python**-socketio<5. Well, slap my thigh.

marcobergman, 2021-04-21:
The latest procedure to get rid on this 'unsupported version of the Socket.IO or Engine.IO' problem is the follwoing sequence:
```
sudo pip3 uninstall python-socketio python-engineio flask-socketio
sudo pip3 install "flask-socketio<5" "python-socketio<5"
```

seandepagnier, 2021-04-21:
I wonder if upgrading the javascript socket.io.min.js in pypilot would
work with the latest socketio and avoid forcing specific python
package versions.

sailoog, 2021-04-22:
stuck to this thread...

marcobergman, 2021-05-01:
dependencies.py still yields the wrong version of flask-socketio; see attached.

[dependencies.py-flask-socketio.txt](https://github.com/pypilot/pypilot/files/6410419/dependencies.py-flask-socketio.txt)

seandepagnier, 2021-05-02:
What system is this on?  It seems broken that you need to use the
--ignore-installed flag to make it work.

marcobergman, 2021-07-24:
Issue seems to be resolved, issue closed. Ref https://forum.openmarine.net/showthread.php?tid=3254&pid=20144#pid20144

closed 2021-07-24 by marcobergman