pypilot/pypilot #81: NMEA 0183 Usb to serial CP2102 recognized as gps0

kagouraki, 2021-03-22:
I have a setup that uses a max232 and a cp2102 usb to serial to get the APB sentence to tinypilot.
So to order is :
|Chartplotter| --NMEA183--> |Max232| --Serial--> |CP2102| --USB--> |Raspberry pi zeroW|
The problem is when i plug in the cp2102 to the rasberry, gpsd rename the device to gps0 so the this code is not reading the nmea coming to the raspberry
>   devgpsdevices = []
    for dev in os.listdir('/dev'):
        devicesd = []
        if dev.startswith('gps'):
            path = '/dev/'+dev
            realpath = os.path.realpath(path)
            devgpsdevices.append(realpath)

>   for device in list(devices):
        if devices[device]['realpath'] in devgpsdevices:
            print('serialprobe removing gps device', device)
            del devices[device]

is there a solution ?

kagouraki, 2021-03-22:
i managed to exclude my chip id from the gpsd rules. i comment out the line refering to my chip id in the file 
`/etc/udev/rules.d/60-gpsd.rules`
This is a temporary solution

closed 2021-03-30 by kagouraki

seandepagnier, 2021-03-30:
Yes, this is a difficult issue to correctly detect if the device is to
be used by gpsd or not.  The gpsd rules is the correct fix.

Sean