pypilot/pypilot #83: unsupported version of the Socket.IO or Engine.IO

mgrouch, 2021-03-26:
Mar 26 14:18:49 lysmarine pypilot_web[32329]: The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)

mgrouch, 2021-03-26:
Would be nice to show which one it is using 
and which one (supported) it needs and how to install it

mgrouch, 2021-03-26:
pip3 install "python-socketio<5"

seems fixes it

seandepagnier, 2021-03-26:
Well this is good to know, but I still don't understand what caused
this, or if pypilot_web can be modified to work with the newer
version.

mgrouch, 2021-03-26:
what if you grep source files for "unsupported version of the Socket.IO"
including third part modules? Might be there is some kind of version check.

marcobergman, 2021-04-09:
#79 ;-)

mgrouch, 2021-05-01:
It seems commits from Apr 25, 2021 broke it again. 
it was working before with <5

seandepagnier, 2021-05-01:
It works now with the new version of socket library, did you try it?

mgrouch, 2021-05-01:
Yes I’ve tried with version 5. It doesn’t give error but doesn’t connect from pypilot_web.

mgrouch, 2021-05-01:
This is how it worked before:

with this install script:

https://github.com/bareboat-necessities/lysmarine_gen/blob/master/install-scripts/4-server/02-pypilot.sh

seandepagnier, 2021-05-01:
What do you get from

python3
import socketio
socketio.__version__

I have '5.2.1'

mgrouch, 2021-05-07:
Shows 5.2.1 but pypilot_web can't connect to pypilot.

seandepagnier, 2021-05-07:
What error do you see?   Does pypilot_client connect?

mgrouch, 2021-05-07:
pypilot_client connects. 

There are no errors in logs.

mgrouch, 2021-05-07:
> Yes. No errors.
> can you stop the pypilot_web service and run it directly from the console?
> if pypilot_client connects then the web service should as well.

Yes no errors. But pypilot web wouldn’t connect

seandepagnier, 2021-08-07:
I have updated the dependency script to hopefully get all the correct versions of libraries needed.

closed 2021-08-07 by seandepagnier