pypilot/pypilot #86: New setup.py new errors

marcobergman, 2021-04-07:
This output was taken from a clean openplotter image. It mentions the following install errors:

[...]
mv: invalid option -- 'r'
[...]
  File "build/bdist.linux-armv7l/egg/pypilot/pilots/intellect.py", line 205
    self.accuracy =
                   ^
SyntaxError: invalid syntax
[...]

and runtime:

ImportError: cannot import name 'linebuffer' from 'pypilot.linebuffer' (/usr/local/lib/python3.7/dist-packages/pypilot-0.24-py3.7-linux-armv7l.egg/pypilot/linebuffer/__init__.py)
[...]
  File "/usr/local/lib/python3.7/dist-packages/pypilot-0.24-py3.7-linux-armv7l.egg/pypilot/failedimports.py", line 1, in <module>
    p#!/usr/bin/env python
NameError: name 'p' is not defined
[pypilot_new_setup.py_new_errors.txt](https://github.com/pypilot/pypilot/files/6273243/pypilot_new_setup.py_new_errors.txt)

seandepagnier, 2021-04-07:
Thanks for reporting, I fixed it.

closed 2021-04-07 by seandepagnier

marcobergman, 2021-04-07:
Indeed thx!