pypilot/pypilot #89: pypilot with zeroconf doesn't connect to signalk

mgrouch, 2021-04-07:
pypilot with zeroconf doesn't connect to signalk.

It finds mopidy instead 

signalk failed to read token /home/pypilot/.pypilot/signalk-token
signalk zeroconf service add Mopidy HTTP server on lysmarine._http._tcp.local. _http._tcp.local.

and fails

mgrouch, 2021-04-07:
and if I disable mopidy it finds avnav server instead of SignalK

then only if I disable both mopidy and avnav it finds SignalK

mgrouch, 2021-04-07:
And even then if fails after approval of token request

signalk post {'state': 'COMPLETED', 'requestId': '95afcc50-de9c-40ab-b0f9-2cac3df05110', 'statusCode': 400, 'message': "A device with clientId 'pypilot-42144498383' has already requested access", 'href': '/signalk/v1/requests/95afcc50-de9c-40ab-b0f9-2cac3df05110', 'ip': '::ffff:10.42.0.1'}
signalk request access url http://10.42.0.1:80/signalk/v1/requests/95afcc50-de9c-40ab-b0f9-2cac3df05110

Even though in  SignalK log it is marked as appproved

notifications.security.accessRequest.any.pypilot-42144498383 | {   "state": "normal",   "method": [],   "message": "The device \"pypilot\" has been approved",   "timestamp": "2021-04-07T21:31:31.108Z" }
-- | --

mgrouch, 2021-04-07:
Request was issued and approved while this was happening pypilot issued another request which gets rejected automatically

seandepagnier, 2021-04-07:
well thank you.   Clearly you are using this in a different environment than openplotter which is fine, the code needs a few adjustments.

marcobergman, 2021-04-08:
See also slack https://signalk-dev.slack.com/archives/C03F1MKQG/p1617828695183000 
"It [pypilot] should also support hard coding the server in case there are multiple."

mgrouch, 2021-04-08:
signalk.py 
should have this 

browser = ServiceBrowser(zeroconf, "_signalk-http._tcp.local.", listener)

I've tested with this fix and it works fine

seandepagnier, 2021-08-07:
I made a bunch of updates to the signalk handling.  I tested on bbn  (derivative of lysmarine) with success with multiple pypilots sharing data.    Could you see if this change is required?  as it now tests for 'swname' to be 'signalk-server'

mgrouch, 2021-08-07:
BBN had already a fix in install scripts 

See https://github.com/bareboat-necessities/lysmarine_gen/blob/master/install-scripts/4-server/02-pypilot.sh

sed -i 's/_http._tcp.local./_signalk-http._tcp.local./' "$(find /usr/local/lib -name signalk.py)" || true

That fix worked for me also. 

Thanks

closed 2022-01-10 by seandepagnier