pypilot/pypilot #92: hat.conf: stateless actions for autopilot control lost

marcobergman, 2021-04-11:
Once you have lost the stateless action for autopilot control from your hat.conf, there seems to be no way to recover them. Removing hat.conf recovers the default_actions, but not the stateless ones.

seandepagnier, 2021-04-11:
there are no default stateless actions...

marcobergman, 2021-04-11:
I'm talking about these: I cannot see them anymore https://github.com/pypilot/pypilot/blob/13d493657c281ef38c39699cd5a596ba3c461d43/hat/hat.py#L255

seandepagnier, 2021-04-12:
It was a small bug I corrected now.   thanks for the report.

closed 2021-04-12 by seandepagnier

marcobergman, 2021-04-12:
I now get "web failed to run process: module 'web' has no attribute 'web_process'" again, this is because the import of the web.py does not work. 

You changed `from . import web` into `sys.path.append(os.path.dirname(os.path.abspath(__file__)))`, but that does not work on a clean openplotter install for some reason, see attached output, where I show the before and after of the sys.path.append. Why was `from . import web` changed? Changing it back into this would solve this problem.

Also because the timout is set to 0 when not on tinypilot, this error runs almost continuous, leaving no cpu for other work. Better  set a default timeout for like 3 seconds, so that when anything goes wrong in the starting of the web server, the ramifications are not that bad.

```
pi@openplotter:~ $ pypilot_hat none
have gpio for raspberry pi
loading config file: /home/pi/.pypilot/hat.conf
failed to load /proc/device-tree/hat/custom_0 : [Errno 2] No such file or direct                                                                                                                                                             ory: '/proc/device-tree/hat/custom_0'
overriding driver default  to command line none
arduino process on  1413
renice: failed to set priority for 1413 (process ID): Permission denied
warning, failed to renice hat arduino process
No hat config, arduino not found
1414 (process ID) old priority 0, new priority 19
web failed to run process: module 'web' has no attribute 'web_process'
1414 (process ID) old priority 19, new priority 19
web failed to run process: module 'web' has no attribute 'web_process'
1414 (process ID) old priority 19, new priority 19
```
[pypilot_issue#92_fix_failed_web_import.txt](https://github.com/pypilot/pypilot/files/6294121/pypilot_issue.92_fix_failed_web_import.txt)

seandepagnier, 2021-04-12:
, your "from . import web" broke my system which worked before the change.   I get:

attempted relative import with no known parent package

So do you have a module "web" or something??
Which openplotter image are you using?

marcobergman, 2021-04-12:
2020-12-16-OpenPlotter-v2-Headless-img.zip
https://cloud.openmarine.net/s/Yapesa2XPJptgaz
After install I do
```
cd 
git clone https://github.com/pypilot/pypilot
cd pypilot
sudo python3 setup.py install
```

then 
```
pip3 install "flask-socketio<5"
pypilot
```

in another terminal
```
pypilot_hat none
```