pypilot/pypilot #96: Control of Stellarium with IMU needed

mgrouch, 2021-04-17:
Control of Stellarium with IMU needed.

For now I implemented it with a hack to read heading from SignalK.
Which is 1 per second data feed for heading.

But it would be nice to make it smooth just as in all smart phone augmented reality apps.
There is an API to control stellarium.
Basically 

select to read location from GPS (GPSD daemon).

Enable RemoteControl plugin in Stellarium:

Edit ~/.stellarium/config.ini

set

[RemoteControl]
autostart = true
...

[plugins_load_at_startup]
RemoteControl = true
...

To control view using heading we can issue the following command:

curl -X POST -d 'az=3.14' http://localhost:8090/api/main/view

az is azimuth in radians.

My hack so far is just:
```

#!/bin/bash

# hack to control stellarium azimuth by true north heading from signalK

stellarium "$@"  &
sleep 5

while [[ -n "$(pidof stellarium)" ]]
do
  # magnetic variation
  MV=$(curl -s http://localhost:80/signalk/v1/api/vessels/self/navigation/magneticVariation/value)
  # magnetic heading
  MH=$(curl -s http://localhost:80/signalk/v1/api/vessels/self/navigation/headingMagnetic/value)
  # true heading
  HT=$(echo $MH + $MV | bc)
  # control stellarium azimuth
  curl -X POST -d "az=$HT" http://localhost:8090/api/main/view
  sleep 2
done

```

seandepagnier, 2021-08-07:
Is this simply because the signalk output rate is too low?

mgrouch, 2021-08-07:
Probably both rate from SignalK is low and rendering in stellarium is not that fast too.

mgrouch, 2021-09-25:
I've improved the solution for that. Stellarium now shows only navigational stars
and satellites and adjusts to heading of the boat, location and time. (every 3 sec, but it's good enough).
It shows horizon and compass scale, ocean landscape. It has on screen angle measurement tool.
I've used it with your celestial navigation plugin to get an idea.
Pretty cool for a demo.

closed 2021-10-03 by mgrouch