pypilot/pypilot #99: Feature Request - Add 6 gain save slots

Rudder2, 2021-05-25:
I really like your TinyPilot computer and all the work you did to make it work well.  It's a great system. 

In-order to make it more user friendly when my father uses my boat it would be nice if there was 6 save slots with away in the web interface and on the OpenCPN Plugin to select them.  I figure with 6 slots I could create a gain profile for about every condition that would be needed.  I'm planning on making a laminated card with the settings but my 76 year old dad said that would be too complicated.

Thank you again for a great project!  I'm about to take my first long trip with the TinyPilot as my 2nd helmsman.  Really look forward to it.

seandepagnier, 2021-08-07:
I think this is eventually coming.   I am adding many more pilot algorithms which each have different features, but what you are asking for is to have multiples of the same pilot.   This is not difficult to implement but I am pondering the best way to interface, by adding additional copies of the same pilot or having slots which save states and if so, what values in addition to gain would be stored in each save slot.

Rudder2, 2021-08-07:
Thank you for consideration of this feature request. I was thinking of the
gains because I haven't been changing other setting past initial setup.

Haven't experimented with more than gains so far because it took gains so
high to keep my boat on course that I'm starting to think my 1980 ram at
the quadrant that came with her is very slow for the displacement but don't
have the funds right now to do anything about that. I need to time how long
it takes to go hard over. Haven't done that yet but off the top of my head
I think it's like 3 - 5 seconds from center to hard over.

I'm looking forward to your machine learning and hope this will start
limiting the gain tweaks.

seandepagnier, 2021-08-07:
3-5 seconds meaning 6-10 seconds for full range.   Are you sure?   Typical setups are 6-8 seconds or even slower on heavy slow boats so this should be ok in many situations.   Are you sailing in very difficult conditions such as waves against current or is the boat unbalanced?   Is it difficult to steer manually?     Did you manage to get it to work at all?   There is a servo.gain parameter which can be used to increase all of the gains, I would try setting it to 1.5 or 2 and compare the difference.       Do you have more information about what gains are required and how they change with conditions?    Any chance to post a video?

Rudder2, 2021-08-07:
Maybe I should open a new topic as to not populate feature request with
this problem.

Rudder2, 2021-08-07:
Opened https://github.com/pypilot/pypilot/issues/100 to discuss no feature
request gain problems.

seandepagnier, 2022-01-10:
I added for now 3 slots (3 copies of basic pilot) as well as potentially a few other pilots.   Could add an option if more are needed, but for now see how this goes.

closed 2022-01-10 by seandepagnier