pypilot/pypilot #100: Really High Gains for caLm weather use

Rudder2, 2021-08-07:
I'm opening this to leave the feature request open for that discussion.

It takes gains so high to keep my boat on course that I'm starting to think my 1980 ram at the quadrant that came with her is very slow for the displacement but don't have the funds right now to do anything about that. I need to time how long it takes to go hard over. Haven't done that yet but off the top of my head I think it's like 3 - 5 seconds from center to hard over.

[image]

This works good for calm water. No fallowing or quartering seas. I think it works with a beam or off the bow and bow sea but can't remember. She steers easily by the wheel but when a quartering sea wave knocks her off I need an immediate 1/4 to 1/2 turn of the wheel to keep her on course depending on power of wave. The autohelm takes too long to get her on course and i have to disenguage and take over to prevent going to shallow water or shore.

 I used these settings and she navigated a OpenCPN course from her dock to the canal and even navigated us down the center of the curvy canal threw bridges to our anchorage 8 hours away, with a continuous watch of course. The only problem was the quartering sea at the end of the trip and when a dick power boat applied full power of our bow and knocked her 35 degrees off course with his 3 huge outboards in the canal which the autohelm couldn't put her on course so i took over till and turned the wheel 1/2 turn and she was quickly back on.

The wheel turns 2.5 turns from center to lock 5 turns from lock to lock. I created these gains under power as there was light wind and I need to order sails before i can test under sail as my main and jib are torn (They are original 1978 sails). 

The autohelm is good as long as it in the open ocean and have unlimited room for her to get under control or calm conditions inland. I really like the autohelm but need to learn alot more and try to create more gain profiles for other conditions if its possible with the way she handles under autopilot especially with how high the gains are. Yes, the Autohelm works and steered us threw storms in the open Gulf of Mexico under power with beem wind and seas to bow winds and seas. Fallowing and quartering seas seem to be the problem.  The sails has seen their last sailing from Fort Myers to Key West so under sail testing had to stop.

This is all the info i can think of.

seandepagnier, 2021-08-07:
5 turns is a lot for a wheel to turn.. but I've seen more.   Is the autohelm turning the wheel?   What exact drive unit is it?    I think you would really get an improvement to have a more powerful motor that can turn the rudder faster.    Being able to turn the rudder faster can save power because the quicker adjustment keeps a closer course making overall smaller corrections.   Did you check the slew settings, and min/max speed settings in pypilot for the servo?    Perhaps increasing these will help make it more aggressive.   Make sure min and max speed are both at 100%, and I would increase the slew settings as well.

Did it seem that you would recover to the correct course eventually if you had more sea room?

The machine learning may help but there is no substitute for a faster motor.   Consider trying to manually steer the boat using your motor.   If this is quite easy I would be surprised.    The only other maybe possibility is the sensors you have could have a bad gyro axis.   If this is true, it would not react to D gain much and the performance would be terrible.   I would run pypilot_scope and move the sensors on all 3 axes and make sure the gyros work correctly.

Rudder2, 2021-08-07:
No, its not turning the wheel directly. All I did was replace the 1983 autohelm computer with your computer and motor controller. I installed the motor controller at the quadrant and PyPilot in the cockpit behind Luxan panel to protect from weather. I have to pull a motor drive unit enable handle at the helm and move the wheel back and fourth past center, assuming i made sure it was centered apond disconnect, till i feel the load of the drive motor on the wheel. Then I press the auto helm engage which activates the hydraulics disengage solenoid, wheel spins free when auto helm is engaged this is how it was designed when installed in 1983, the auto helm is the only device in control till I press the deactivate button on the autohelm and then I regain helm control.

The drive motor is a linear Drive motor installed on the quadrant. The old autohelm is an Alpha Marine Systems 3000 Autopilot installed in 1983. 

Yes, if i have enough sea room she would recover. Until I craned the gains up that high she looked like a drunk sailor meandering back and fourth 6 feet on either side of course. 

I tried one gained at a time then tried combinations till i came up with what's in the picture. The PyPilot swears she isn't using very much power when in operation... I know she is using far less than the original computer even with the high gains. The Hydraulics Disable solenoid uses 2 Amps while autohelm is engaged but the motor controller doesn't take that in to account in the power usage calculation. I have the hydraulics disable solenoid connected to the clutch output on the motor controller.

I haven't felt comfortable trying to manually control her with the controls as the latency in the WiFi is nerve racking and i can't see my quadrant to know what the rudder is doing. Yes, there is a rudder position sensor built in to the motor drive unit and i did calibrate it. When testing I will disassemble the aft bunk so i can see the quadrant threw the aft hatch and know more what she is doing. D gain does make a difference I can tell. D derivative gain made a huge difference when seas were bigger. 

I plan on installing hardware buttons on the pedestal to be able to control the autohelm especially since i need to be able to disengage fast to regain helm control. If I flip the PyPilot computer off will that instantly give me helm control? Been to scared to try it because if that doesn't work I would have to jump below to main breaker panel to kill Autohelm Breaker which goes to the motor controller at the quadrant. When i install these buttons I will feel way more confident trying to control the boat manually but, it will only work if the PyPilot sends the clutch signal so the hydraulics are disabled.

I know I should have a hydraulic rudder control motor but the boat didn't come with one and they are expensive. I will buy one in the future.

I doubt it is a computer and/or sensor problem but it might be. The gains change color and seam to make a big difference just need to be very high.

I will try changing the slew and servo.gain settings.Yes Min/Max are 100%.

seandepagnier, 2021-08-08:
The new software in motor controller supports pwm on clutch output to save some power.  Also supports rf wireless remote controls to make it easier to use buttons as they can be wireless. 

 Turning the computer off the motor controller will time out within a few seconds and release, you could test this and see.   If you could instead turn the motor controller off, or add a manual switch to disconnect the clutch may be a good idea in your situation.

As for manual control lag, I recently improved this significantly from up to 250ms lag down to 20-30, but I have not released an updated image yet.

As for using less power than the old pilot.. is it from not moving as often?   Did the old pilot steer better?   Can you increase the gains even more?   Try with I set to 0, and also DD up to 1.5 times D.   If you give more feedback it will help me more. 

As for not seeing what the rudder is doing:  can you display the rudder output in opencpn or some program with a nmea connection to pypilot?   It should output rudder angle nmea sentences as well as over signalk.

Requiring such high gains indicates that the drive motor is slow to correct the boat movements.   How much current does it draw while moving the rudder?  Would it be possible to somehow attach the motor arm closer to the rudder shaft thereby reducing the leverage and increasing the speed?   The current will increase, but  if it can handle a bit more you could increase speed this way.   I would also ensure the voltage drop by the time it reaches the motor is not very high.

Rudder2, 2021-08-08:
The voltage drop as reported by PyPilot statistics is 1.5 - 2 volts based on what the motor is doing. I can remove 15 feet of wire from the wire that goes to the quadrant. I hooked it up like it was installed from the manufacturer installer of Alpha Marine. The system with the motor controller being at the quadrant It's 15 feet closer to run the motor controller directly from my breaker panel and this is on the to do list.

The motor controller and quadrant is 15 feet from the cockpit and 25 feet from where the original computer and PyPilot is mounted is why i have the motor controller at the quadrant. Running the wire from there to the pedestal is a PITA and was hoping to only have the data and power wire running back there. Now, adding a switch from the motor controller to the pedestal might be doable but that would take 6 feet more wire. I'm trying to shorten the power wire going to the motor controller because of the power drop.

I think the less power comes from the newer electronics and the PyPilot moves the rudder waits than moves it again. This is how i steer but the slow motor response makes this not good because the computer waits to long and doesn't move the rudder more than the last time when it didn't work. I know my boat and can almost predict the rubber movements the computer is show at this which might be a slow drive motor problem.

Your right, i forgot to turn DD up to 1.5 times D the last gain update. I don't remember why it helped and i will try DD 1.5 and I 0 also next time in addition to previous suggestions.  

I haven't figured out how to display the rudder angle on OpenCPN yet. I'm trying to learn this whole open source system as i go. I still like it. OpenPlotter is my primary navigation system but didn't want one computer to run everything so opted to have a separate PyPilot computer so if navigation system went down i still had autopilot which was wise when i had a leak and the OpenPlotter computer got wet.

The way my quadrant is set up i can't move the rudder drive motor. I need to measure the current drawn but it seams very low based on the wattage estimate. Locked motor current is 8 Amps. I remember that. I set PyPilot to 7.5 Amps as the stop point and that seams to so well.  The motor controller stops because of rudder position not Amps.

seandepagnier, 2021-08-08:
1.5 -2 volts is too much.   It may momentarily drop when first moving, but during motion the drop should be 0.3 to 0.2 volts.   If you do a basic calculation (not exact) you could be losing ~ 40% of the power if you really drop 2 volts.  Consider using thicker gauge wire from battery to controller as well as controller to drive.   Also measuring with a volt meter the voltage output of the controller when moving would be useful to verify it.

You could maybe use a relay to cut power to the solenoid or just wire the switch in series with the solenoid rather than the controller to avoid heavy wire.   Normally not an issue, but it is very important to ensure you can take manual control even if the electronics or software makes a mistake.

As for not moving enough..  You can adjust servo.period, possibly reducing it.  The default is 0.4 but try lower values of 0.3 or 0.2 could give a faster reaction usually at the expense of power consumption.    Also just increasing the gains even more could maybe help as well as playing with servo.gain (as mentioned) to values of 1.5 or even 2

As for the drawn power being very low, this makes sense, but indicates you could change the mechanical advantage.   I don't know your exact setup, but really you should consider this as it would gain a lot by reducing this so that the same motor movement moves the rudder further.   For example attaching the drive motor to a different pin on the quadrant, say 60% of where it is now could have a large effect.    Keep in mind most people have success with gains 1/4th the values you are using.   It also may be that your rudder is not very large?

Rudder2, 2021-08-08:
Yes, I agree, 1.5 - 2 volts drop is way to much. The first attempt at resolving this is going to be to take 15 feet out of the wire run. I shouldn't of assumed the manufacturer install tech installed the wire gauge correct for this run. The PyPilot computer has no power issues. Controller is 2 feet from drive motor.

The hydraulics Disable Solenoid is 6" from the motor controller. Motor controller is 15 feet from from pedestal it would take 20 feet of wire to get from the quadrant to the pedestal. She is a center cockpit. Yes, the hydraulics being hard to re-enable makes me nervous. The fraction of a second for the PyPilot to respond to the wireless button is nerve racking at times.

Ok, will add servo.period to the list of things to try. I'm not as concerned with power consumption as I have a big house bank and will have 1200 watts of solar before I move onboard.

I'm unable to move the drive unit because of the way the drive unit disconnect is installed to the quadrant and clearance from the bunk to the quadrant requires mounting everything to the bottom and the drive motor disconnect plate is mounted to the top of the quadrant as it relies on gravity and one huge bolt to attach it. I really wish i could move the drive motor closer to the rudder shaft. My rudder is huge also. On the forms people refer to our rudders as a barn door. The rudder is 4.5 feet deep, attached to bottom of keel, and 4 feet long.  Once i get her balanced it normal only takes an inch or two of wheel movement under sail and power with only big rudder movements if a wave knocks her off course. Wind doesn't effect her very much because of her long keel.

S/V Meandering Abode is 41 feet long, 14 foot beam, full keel, 27000 lbs, and lots of windage. I've read the blog post of the steel boat you tested PyPiloy on before I made up my mind and know it's a capable system. Getting a hydraulic autohelm pump will be the end all future fix of all problems but that's not currently in the budget. Will be hopefully in 2 - 4 years but Sails are higher priority.

closed 2022-01-10 by seandepagnier