pypilot/pypilot #105: install error with latest version

mgrouch, 2021-09-03:
Cloning into 'pypilot_data'...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 16 (delta 0), reused 13 (delta 0), pack-reused 0
Unpacking objects: 100% (16/16), done.
'./pypilot_data/README.md' -> './pypilot/README.md'
'./pypilot_data/arduino/weathersensors/font.ttf' -> './pypilot/arduino/weathersensors/font.ttf'
'./pypilot_data/hat/font.ttf' -> './pypilot/hat/font.ttf'
'./pypilot_data/ui/Vagabond.mtl' -> './pypilot/ui/Vagabond.mtl'
'./pypilot_data/ui/Vagabond.obj' -> './pypilot/ui/Vagabond.obj'
'./pypilot_data/ui/compass.png' -> './pypilot/ui/compass.png'
'./pypilot_data/web/static/font/Roboto-Regular.ttf' -> './pypilot/web/static/font/Roboto-Regular.ttf'
/install-scripts/stageCache/pypilot /install-scripts/stageCache /install-scripts
/usr/bin/apt
checking for RTIMULIB2...Traceback (most recent call last):
  File "setup.py", line 20, in <module>
    import dependencies
  File "/install-scripts/stageCache/pypilot/dependencies.py", line 196, in <module>
    s.install()
  File "/install-scripts/stageCache/pypilot/dependencies.py", line 140, in install
    if dep.test():
  File "/install-scripts/stageCache/pypilot/dependencies.py", line 108, in test
    from importlib.metadata import version
ModuleNotFoundError: No module named 'importlib.metadata'

Exited with code exit status 1

CircleCI received exit code 1

Thanks

seandepagnier, 2021-09-03:
Thanks for testing this.   What system are you running?   Is it openplotter?

Can this fix this with:
sudo python3 -m pip install importlib
?

mgrouch, 2021-09-03:
BBN lysmarine OS.

I've tried sudo python3 -m pip install importlib
and 

sudo python3 -m pip install importlib-metadata

Both do not fix the issue.

Thanks

mgrouch, 2021-09-03:
if I change dependencies.py

from importlib.metadata import version

to 

from importlib_metadata import version

only then it works.

Thanks

seandepagnier, 2021-09-04:
wow!!

what python --version?

Is this just the latestest bbn version?

mgrouch, 2021-09-06:
Yes it is latest BBN version. 
python3 version 3.7.3

closed 2021-09-25 by seandepagnier