pypilot/pypilot #107: make services installation optional step

mgrouch, 2021-09-14:
make services installation optional step

Current code hard codes user  'pi' in service definitions.
Some systems like: 

https://github.com/bareboat-necessities/lysmarine_gen

do not even have user 'pi'

Thanks,
--MG

seandepagnier, 2021-09-18:
ok fair enough I will push this bit into the openplotter specific package

closed 2021-09-18 by seandepagnier

mgrouch, 2021-09-18:
I hope you had a chance to download and evaluate BBN lysmarine image.
I’d be happy to hear what are your experiences. All your off shore sailing software weather routing, climatology, weather fax, grib plugin, etc are included. Thanks for making those. Without you these features possibly would never materialize. 
Please take a closer look at BBN lysmarine OS. It’s completely open source and built continuously on each git push by circleci. Any issues you find I’d be glad to hear about. 

Thanks