pypilot/pypilot #108: starting pypilot service breaks bluetooth

mgrouch, 2021-09-24:
starting pypilot service breaks bluetooth

If I start pypilot service I start getting errors described here:

https://github.com/raspberrypi/linux/issues/4601

without pypilot service bluetooth works.

Thanks,
--MG

mgrouch, 2021-09-24:
Is this probing logic of ttyAMA0 breaks it?

mgrouch, 2021-09-24:
The /dev/ttyAMA0 previously used to access the UART now connects to Bluetooth.
The miniUART is now available on /dev/ttyS0.
In the latest operating system software there is a /dev/serial0 which selects the appropriate device so you can replace /dev/ttyAMA0 with /dev/serial0 and use the same software on the Pi3 and earlier models.

Looks like instead of referring to it as AMA0 it needs to refer to it as serial0. 

Thanks

mgrouch, 2021-09-25:
I've did this

```
sudo sed -i 's/ttyAMA0/serial1/' "$(find /usr/local/lib -name serialprobe.py)"
sudo sed -i "s/'ttyAMA'//" "$(find /usr/local/lib -name serialprobe.py)"

```

And bluetooth started working but sometimes only.
It seems to me probing code still tries to probe bluetooth device thinking it is a serial port it needs to use.
Propably via another path on /dev/ file system.

This probing of bluetooth device really breaks it.

seandepagnier, 2021-09-25:
pypilot must have access to /dev/ttyAMA0 and so bluetooth must not be using this port.

The mini uart is broken and cannot work reliably with pypilot.

It may be possible to use the miniuart (/dev/ttyS0) with bluetooth but considering it is unreliable with pypilot it probably is not useful for bluetooth either.   I find reports online of people constantly restarting hciattach when it fails to work around the buggy drivers/hardware there.    If you want to use bluetooth I suggest using a usb bluetooth device, or using one of the alternative serial ports of the pi4 (/dev/ttyAMA[1-4]) for pypilot but these have never been tested yet.

   pypilot probes if you do not configure explicit serial ports.   If you add devices to ~/.pypilot/serial_ports it will only use those ports.   Alternately if you add devices to ~/.pypilot/blacklist_serial_ports it will not use those ports.

The openplotter-pypilot plugin configures these ports, but since we decided not to use this, then it is up to the OS  in question to handle configuring which programs can access which serial ports and I do not know what your plan is for handling this, but to configure pypilot it would just need to write to those files.   pypilot does negotiate with gpsd, but currently does not negotiate with other programs.   I am considering negotiating with signalk server to automatically not probe devices configured there but have not worked it out yet.

mgrouch, 2021-09-25:
/dev/ttyAMA0 is the name of bluetooth port on rpi4

mgrouch, 2021-09-25:
Does pypilot need to use onboard UART? In case of not tinypilot configuration?

mgrouch, 2021-09-25:
What is the syntax of ~/.pypilot/blacklist_serial_ports

Comma delimited one line or one port per line?

Thanks

mgrouch, 2021-09-25:
My plan for pypilot on BBN OS is to be a source of heading, compass and heel, pitch.
Attaching servo to BBN OS and rudder motor is not a good idea anyway and should be discouraged and not supported.
It should be just like a proxy to tinypilot box. tinypilot box what needs to be actual autopilot.
So I'm trying to find a way for pypilot on BBN OS not to conflict with other programs over devices.
Because on BBN OS pypilot should not use any devices or worst case IMU only.
The point is to give users stable UI in BBN OS and promote possibility of connecting tinypilot.
Actual control of the rudder motor and IMU readings should be done on your hardware.
So the use cases I'm shooting for:

1. User installs BBN OS . Sees pypilot component it in. Gets familiar with UI part of it. Wants to install autopilot.
   -> Solution Use familiar UI on BBN OS but actual autopilot hardware is tinypilot.

2. User installs BBN OS . Sees pypilot component it in. Gets familiar with UI part of it. Wants to add compass and gyro,
  i. e. IMU only. Puts it as a hat on same pi. It's i2c. No serial ports taken.

These are only 2 use cases I want to have. BBN OS is UI and not a device to run autopilot itself on.

So looks like I need to black list all serial ports for pypilot on BBN OS.
Can I use * in ~/.pypilot/blacklist_serial_ports?

What do you think?
Thanks,

mgrouch, 2021-09-25:
Thanks for  ~/.pypilot/blacklist_serial_ports tip.
With it I've managed to get pypilot to do I what I needed.
No conflicts with bluetooth. It's now pretty much works as pypilot_boatimu, not
opening serial ports.

closed 2021-09-25 by mgrouch