pypilot/pypilot #109: pypilot_boatimu is a code duplication

mgrouch, 2021-09-25:
pypilot_boatimu is a code duplication.

It would be better to have just an argument to pypilot to start it in 'IMU only' mode.
There is no need to create duplicated code and do extra maintenance.

Thanks

seandepagnier, 2021-09-25:
It is not a code duplication just an alternate entry point.  

For example there is technically a "pypilot_nmea" which only receives nmea data and distributes it as a multiplexer but it is not exported as an official entry.   There is similarly "pypilot_signalk" (testing signalk support) "pypilot_servo" (testing motor controller without using imu) etc.  but these are just used in development/testing,  the pypilot_boatimu is the only one that was found to be used by end users.

From a distribution standpoint it may be easier to deal with a command line argument, is this what you are suggesting?   I can see what you mean if this is the case, but it is not extra maintenance for me just a different paradigm.   It seems to me you find this frustrating, and I don't fault you but I offer you an explanation.   I don't really mind making concise and intuitive changes here so I value feedback.

Do you suggest it is easier for BBN/lysmarine etc?   How will the user select the commandline arguments to the services?   Currently they could use systemctl and enable/disable pypilot vs pypilot_boatimu but I can see how this is confusing too.   So how will they set this mode up?    Or are you suggesting that they always run the default pypilot service (in which case you could just ignore the fact that pypilot_boatimu exists) ?

mgrouch, 2021-09-25:
pypilot_boatimu doesn't seem to have code to request a token from signalk though.
Anyway. I do not need pypilot_boatimu. With banning the serial ports from pypilot,
regular pypilot as good as pypilot_boatimu (even better).

Thanks

closed 2021-09-25 by mgrouch