pypilot/pypilot #124: Suggestion for german translations webinterface pypilot

Moeritsen, 2022-01-25:
as already discussed in the openmarine-forum, i would like to make a suggestion for an adapted translation of many terms of the web interface into german. in the pdf you can see which terms could be replaced where
[possible PypilotWebTranslation-German.pdf](https://github.com/pypilot/pypilot/files/7934045/possible.PypilotWebTranslation-German.pdf)
.

seandepagnier, 2022-01-26:
I am sorry I do not understand german, but I am at a conflict as another german disagrees with you on some translations:
https://github.com/pypilot/pypilot/commit/c766b9ffebe65630f31563448262c4137a4f73e1

Please review this commit and let me know what you think.  Perhaps we can get his opinon/

if possible it is better to edit this file:
https://github.com/pypilot/pypilot/blob/master/web/translations/de/LC_MESSAGES/messages.po

Otherwise you will miss strings that were not displayed at that time.

Moeritsen, 2022-01-26:
I have commented on the file and will contact S-I-M-O-N to discuss the translations. There should be a number of other terms. in which file are they?

seandepagnier, 2022-01-26:
At this time the translations are split in several locations:

https://github.com/pypilot/pypilot/blob/master/web/translations/de/LC_MESSAGES/messages.po
https://github.com/pypilot/pypilot/blob/master/hat/translations/de/LC_MESSAGES/messages.po
https://github.com/pypilot/pypilot/blob/master/hat/locale/de/LC_MESSAGES/pypilot_hat.po
https://github.com/pypilot/pypilot/blob/master/pypilot/locale/de/LC_MESSAGES/pypilot.po

closed 2024-01-07 by seandepagnier

rgleason, 2024-01-07:
Sean, why not use the normal crowdin translation process?