pypilot/pypilot #125: pypilot doesn't give NMEA rudder angle on port 20220

mgrouch, 2022-02-09:
pypilot doesn't give NMEA rudder angle on port 20220

Thanks

seandepagnier, 2022-02-09:
Are you sure pypilot is getting rudder angle?

mgrouch, 2022-02-09:
Oh. It sends this sentence only if it has it from motor controller?
Thanks

seandepagnier, 2022-02-10:
I think so.   How else is it getting rudder feedback?

pypilot generally doesn't relay data, but it does translate things.
It may send it if it receives over a serial port via nmea too, but for
the case of tcp or signalk, you could just get the data directly from
that source and there is no real need for it to go through pypilot
correct?

mgrouch, 2022-02-14:
Yes. Pypilot shouldn’t relay other data. I was talking about a case when only pypilot knows rudder position from tinypilot sensors. 

Will it generate NMEA sentence for rudder position on port 20220?

or it just uses rudder position value for autopilot functionality but doesn’t act as NMEA talker of rudder position sentence?

thanks

mgrouch, 2022-02-14:
Ok. Thanks. Then it is the behavior I’d expect.

closed 2022-02-14 by mgrouch