pypilot/pypilot #126: Barometer and temperature

mgrouch, 2022-02-09:
Due to nature of most IMU modules being designed for drones they would have barometer and thermometer.
There are NMEA sentences for barometer and air temperature.
Why not to read those values and send it as NMEA for a recognized IMU?
Barometer values are important because many apps (SignalK, OpenCPN) have barographs built in (Traditional weather predictor).

Some IMUs have even humidity sensor. I'm not sure NMEA sentence exists for humidity though. 
But it can be used for fog conditions detection.

Just food for thoughts on improvements.

Thanks

seandepagnier, 2022-02-09:
Which imu in particular?     If you use openplotter you can get the barometer info out using a separate application.   Currently there is no support for this under tinycore.

mgrouch, 2022-02-09:
I use BBN OS with Waveshare 10 dof v2 (C). It works fine with SignalK plugin.
No openplotter stuff is involved.

I was thinking about people who use just tinypilot OS.

seandepagnier, 2022-02-10:
There could be a separate service that pushes the data to signalk that runs on tinypilot.    If pypilot were to handle the i2c directly it somewhat clutters autopilot specific software with something that is not used by the autopilot.   I also support this sensor externally via usb normally using my weather sensors in which case the data gets relayed from usb and pushed out nmea, but not signalk.

So I am not sure the optimal way to incorporate support as there are multiple options and users might want to configure it a bit, and complicating pypilot for no autopilot benefit.   It's not really a priority for me at this time, but I would accept pull requests to tinypilot if they do not cause any issues.