pypilot/pypilot #137: Latest image does not have LCD support

ronangaillard, 2022-05-23:
Hi,

I downloaded and flashed the latest image from here : [tinypilot_2021_11_16.img.xz](https://pypilot.org/download.php?Down=images/tinypilot_2021_11_16.img.xz)

I plugged the Nokia 5150 screen to my raspberry, unfortunately nothing shows up. 

It looks like the LCD folder is missing in this image, that would explain why it does not work.

I manually updated pypilot using git. I was not able to fully start pypilot but graphics would show up on the LCD screen so I know it's correctly connected to the raspi.

One solution I guess would be to update via git as I did but it's not working well for me (I can post logs once I get back home).

In the mean time I would like to know if it's normal that the LCD screen is not working with the 2021_11_16 image which seems to be the latest.

Thanks,

Ronan

seandepagnier, 2022-05-23:
no its not normal, it should work.

Did the image maybe get corrupted?

ronangaillard, 2022-05-24:
Here is the md5 of my extracted image : `bbc7d5280bea13073baf74a64f483cf0` (tinypilot_2021_11_16.img)

I don't think it could be corrupted as everything else seems to work perfectly (mostly tried web UI + SSH).

Is there any config file I should change to "enable" the screen in software ?

seandepagnier, 2022-05-25:
Does it work with the next older image?
https://pypilot.org/images/

If so,  you just need to ssh in and:
sudo sv d pypilot_hat
nano ~/.pypilot/hat.conf

edit to fix the screen and reboot.

ronangaillard, 2022-05-31:
Sorry for the delay I was not home.

I had a look at ` ~/.pypilot/hat.conf` and indeed the lcd driver was not set to `nokia5110`. Setting it to the good value fixed it, and now I have graphics displaying on my screen :) 

But I just have horizontal and vertical lines moving across the screen, but maybe this. is due to the fact that my mpu9250 is not well recognized.

I'm closing this issue as the bugs I have now are not linked to the issue I created.

closed 2022-05-31 by ronangaillard