pypilot/pypilot #140: Feature request: NMEA sentence VHW and HDG as output from pypilot when it recieves GPS input (RMC)

peterdrost, 2022-06-05:
I would like pypilot to transmit nmea-sentences HDG and VHW when it gets variation data from GPS and speed from log.:
=== VHW - Water speed and heading ===
------------------------------------------------------------------------------
                1  2 3   4   5  6  7  8  9
                 |   |  |   |     |   |   |   |   |
 $--VHW,x.x,T,x.x,M,x.x,N,x.x,K*hh<CR><LF>
------------------------------------------------------------------------------
Field Number:
1. Degress True
2. T = True
3. Degrees Magnetic
4. M = Magnetic
5. Knots (speed of vessel relative to the water)
6. N = Knots
7. Kilometers (speed of vessel relative to the water)
8. K = Kilometers
9. Checksum

seandepagnier, 2022-06-11:
What is this used for?    You are using gps declination and water speed??   What about currents?

peterdrost, 2022-06-11:
Hi Sean
I've got an app that uses these two sentences for calculating lay-lines and
it can't use HDM and can't parse my log's VHW, as it only holds the water
speed. Having pypilot making these sentences spares me from having a second
system running opencpn to make them.
So its a nice to have feature, but it will make my setup just a bit les
complicated.
Regards
Peter

mgrouch, 2022-07-20:
I think this feature is not needed and it goes against what be best practices. 
you can easily make a multiplexer with kplex on raspberry pi or even on OpenWrt router. 

It can take input from pypilot and GPS and mix it together in one tcp stream.

seandepagnier, 2022-07-21:
I am implementing filters to combine more data types and compute this, but it is not working yet.

   It would then make sense for pypilot to output there as it would use the gyros and accelerometers to compensate the output giving much higher data rate than opencpn is capable of.

mgrouch, 2022-07-21:
It’s important to not create data loops with all these NMEA set ups

seandepagnier, 2024-01-07:
this feature is essentially implemented but not tested as a separate kalman filter that blends the inertial sensors with the gps for high rate position and heading.

closed 2024-01-07 by seandepagnier