pypilot/pypilot #141: Undesired action: pypilot goes to GPS mode when recieving routing data without any warning

peterdrost, 2022-06-11:
I have found that when in autopilot mode (compass, wind, true wind) and you start transmitting route-data from your plotter to pypilot, it automaticly swithes to GPS mode and it is not possible to switch to another mode.
I would very much like pypilot only to switch to a mode when you actively selects it. 
I sail in narrow and very busy waters and this behaviour poses a risk, eg. if you select the wrong waypoint, pypilot will without any notifications turn the boat around.

seandepagnier, 2022-06-11:
you can change this behavior in opencpn anyway by changing the talker ID to GP instead of default EC.

seandepagnier, 2022-06-11:
perhaps using the forward nmea option (In the plugin) to avoid connection setup and autodetect pypilot, there should also be an option for this.   Currently you have to manually make the 20220 connection and change the sender id

peterdrost, 2022-06-12:
Hi Sean
I'm not using opencpn as my plotter.
My setup consists of a wifi accespoint that handles all my wifi units
including my tinypilot/pypilot and a YachtDevices YDNR-02
nmea0183/2000/seatalk1 multiplexer/server.
Pypilot is connected to the YDNR-02 on port 20220.
My chartplotter, Lowrance hds7m gen2, is connected to the YDNR-02 with
nmea2000 and all messages is transmitted as nmea0183 on port 20220 with the
identifier YD. I can change the identifier on the YDNR-02 if need be.
Regards
Peter

seandepagnier, 2022-06-12:
If you can configure your identifier to GP for the APB message.

Check the nmea stream and you should find GPAPB to get the behavior
you are after

peterdrost, 2022-06-22:
I've got the YDNR-02 to transmit NMEA 0183 with the GP header and it works :-)
Regard 
Peter

closed 2022-06-22 by peterdrost