pypilot/pypilot #142: command line utility to read pypilot_hat config parameters from firmware

mgrouch, 2022-08-03:
Something like:

pypilot_hat_conf  

-check 
    verify that hardware hat is installed
-lcd
   verify that hardware hat has lcd and display its config (parseable by shell)
-lirc
   verify that hardware hat has IR and display its config (parseable by shell)
-mpu
   verify that hardware hat has MPU and display its config (parseable by shell)

so it can be used by other shell scripts if needed.

Thanks

seandepagnier, 2022-08-05:
Trying to understand where you are going with this.  Are those switches like -lcd?   I can make that, but what would invoke this and why are the various switches needed?

What about a script
pypilot_config_hardware

This simply runs, detects everything, makes any needed changes.   The pypilot_detect service in bbn just runs it on boot.    What other functions am I missing?

mgrouch, 2022-08-05:
I didn’t know pypilot_config_hardware
exists. Is it documented somewhere?

Can we invoke it from service and how?

thanks

seandepagnier, 2024-01-07:
Are we done with this?    The pypilot hat's all have an eeprom that creates the path /proc/device-tree/hat/custom_0 which contains the information about the hardware on that hat.

closed 2024-01-07 by seandepagnier