pypilot/pypilot #143: Exception converting signalk->pypilot

marcobergman, 2022-08-16:
These exceptions occur both in pypilot 0.24 and 0.34, against a Signal K Server version 1.44.0.

Might be related to https://www.cruisersforum.com/forums/f134/opencpn-5-6-2-signalk-connection-causes-freeze-265999-3.html#post3647067

> Exception converting signalk->pypilot unsupported operand type(s) for /: 'NoneType' and 'float' {'GP': {'navigation.position': {'longitude': 5.2561585, 'latitude': 52.0780525}, 'navigation.courseOverGroundTrue': None, 'navigation.speedOverGround': 0.03704000938346904}}

Exception converting signalk->pypilot unsupported operand type(s) for /: 'NoneType' and 'float' {'pypilot': {}, 'GP': {'navigation.position': {'longitude': 5.256074666666667, 'latitude': 52.078021666666665}, 'navigation.courseOverGroundTrue': None, 'navigation.speedOverGround': 0.051444457477040344}}

seandepagnier, 2022-08-16:
Thanks!

I will be pushing changes in the next day or two.  I have already fixed this.

I think it doesnt matter much since it's only when course is "None" ?

marcobergman, 2022-08-16:
Thx!

I guess the course being None happens when you are at home or in the harbor or so. But I'm not setting sail when I have an error ;-)

marcobergman, 2022-08-22:
Just an fyi, related signalk issue https://github.com/SignalK/nmea0183-signalk/issues/226

marcobergman, 2022-08-23:
I did not seem to be able to replicate it with a freshly compiled 0.34 setup (seet issue #144), so I changed my 0.24 version as follows:

```
                                print ('about to divide {} by {}'.format(value, signalk_conversion))
                                data[pypilot_path] = value / signalk_conversion
                        except Exception as e:
                            print('Exception converting signalk->pypilot', e, self.signalk_values)
                            break
```

and the error message is still the same:

```
about to divide None by 0.017453292519943295
Exception converting signalk->pypilot unsupported operand type(s) for /: 'NoneType' and 'float' {'GP': {'navigation.position': {'longitude': 5.256051666666667, 'latitude': 52.078047833333336}, 'navigation.courseOverGroundTrue': None, 'navigation.speedOverGround': 0.036011120233928245}}
```

which coincides with the message I get when dividing None by a float:

```tc@box:~$ python3 -q
>>> None / 0.017453292519943295
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported operand type(s) for /: 'NoneType' and 'float'
```
so I would think that line https://github.com/pypilot/pypilot/blob/master/pypilot/signalk.py#L364 is the culprit. I've tried as follows and that seems to solve the issue:

```
                                if value is None:
                                    data[pypilot_path] = None
                                else:
                                    data[pypilot_path] = value / signalk_conversion
```

pwagsc, 2022-09-01:
Similar problem on .35 with same fix as Marco.  Also had an issue when attempting to convert 'navigation.position' which passes a dict value (containing lat/lon).  Since the type of pypilot_path is not a dict, there is no attempt to breakdown the lat/lon in the if type ... statement and the code fails because the computation with value as a dictionary is invalid.  Not sure what the intent is here but I don't readily see a need for lat/lon in the autopilot so I corrected that by ignoring it as follows:

                            value = values[signalk_path]
                            if type(pypilot_path) == type({}): # single path translates to multiple pypilot
                                for signalk_key, pypilot_key in pypilot_path.items():
                                    if not value[signalk_key] is None:
                                        data[pypilot_key] = value[signalk_key] / signalk_conversion
                            elif value is None
--> per Marco          data[pypilot_path] = None
--> ignore dict    elif not type(value) is dict:
                                data[pypilot_path] = value / signalk_conversion

Now no crashes

pwagsc, 2022-09-01:
New to github so formatting got messed up.  Should have been following with comments

                            value = values[signalk_path]
                            if type(pypilot_path) == type({}): # single path translates to multiple pypilot
                                for signalk_key, pypilot_key in pypilot_path.items():
                                    if not value[signalk_key] is None:
                                        data[pypilot_key] = value[signalk_key] / signalk_conversion
                            elif value is None:
                                data[pypilot_path] = None
                            elif not type(value) is dict:
                                data[pypilot_path] = value / signalk_conversion

seandepagnier, 2022-09-01:
Thank you for this.   I tried something slightly different in git.

The autopilot does not use lat/lon, but will in the future.    It needs to be aware of these so it can pass the data along and filter it.   For example, it can use the gyros/accelerometer and compass sensors to produce a high rate gps output for better speed and direction estimation as well as including lat and lon for improved plotting and accuracy.

pwagsc, 2022-09-02:
Hi Sean,

Good fix passing on the fix dictionary as value for future use.  I am curious though why you simply didn't add the lat/lon structure to the systemk table so the code just above the fix would break it down nicely into lat and lon data entries instead of passing the fix structure.