pypilot/pypilot #144: signalk bad status, rejecting token

marcobergman, 2022-08-23:
Openplotter 3, signalk 1.44.0, pypilot 0.34 compiled from source:

```signalk see if token is ready http://10.10.10.1:3000/signalk/v1/requests/356ba04f-2b3c-4efb-805d-305ed682f34b {'state': 'COMPLETED', 'requestId': '356ba04f-2b3c-4efb-805d-305ed682f34b', 'accessRequest': {'permission': 'APPROVED', 'token': 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJkZXZpY2UiOiJweXBpbG90LTQzMzEzODM5NDcwIiwiaWF0IjoxNjYxMjUyNTk2LCJleHAiOjE2OTI4MTAxOTZ9.i_7hkYdt0c2XP0Y8xnGvqVeGiKEq4fpfHx-rSHN-lw8'}, 'statusCode': 200, 'href': '/signalk/v1/requests/356ba04f-2b3c-4efb-805d-305ed682f34b', 'ip': '10.10.10.1'}
signalk received token eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJkZXZpY2UiOiJweXBpbG90LTQzMzEzODM5NDcwIiwiaWF0IjoxNjYxMjUyNTk2LCJleHAiOjE2OTI4MTAxOTZ9.i_7hkYdt0c2XP0Y8xnGvqVeGiKEq4fpfHx-rSHN-lw8
signalk bad status, rejecting token
signalk post {'state': 'PENDING', 'requestId': '5c7a3449-d093-4c3d-ac1c-2ad3b0b69240', 'statusCode': 202, 'href': '/signalk/v1/requests/5c7a3449-d093-4c3d-ac1c-2ad3b0b69240', 'ip': '10.10.10.1'}
```

sailoog, 2022-08-23:
A new openplotter-pypilot version is about to be published for openplotter 3. It contains a signal k connection manager to make easy requesting and aprobe pypilot connections. We are sailing right now and it still will take some days.

marcobergman, 2022-08-23:
> A new openplotter-pypilot version is about to be published for openplotter 3. It contains a signal k connection manager to make easy requesting and aprobe pypilot connections. We are sailing right now and it still will take some days.

That's excellent; I recently tried the old version, cloned from github, on openplotter 3, but it did not seem to work yet. A proper solution for this has been long sought for, and the moment it appears to be operationalised I'm happy to revise the pypilot workbook for it.

Although this issue is unrelated to openplotter-pypilot, you might run into it in finishing it off.