pypilot/pypilot #145: dependencies.py and setup.py script are not working properly if run from CI build

mgrouch, 2022-08-27:
There is some logic for detecting Pi, etc which skips installing wiringPi if run inside CI build (dockers)

It can only work on actual pi. 
Builds on pi are very slow. CircleCi nowadays has native armhf/arm64 support which works way faster for builds.

Thanks
--MG

rgleason, 2022-08-27:
Agreed about the builds, can you point us to the native armhf/arm64 support builds?

mgrouch, 2022-08-29:
https://github.com/bareboat-necessities/lysmarine_gen

it builds arm64 bullseye on CircleCi 
natively on arm64 

pypilot part is here

https://github.com/bareboat-necessities/lysmarine_gen/blob/bullseye/install-scripts/4-server/02-pypilot.sh

mgrouch, 2022-08-29:
And CircleCi related stuff here

https://github.com/bareboat-necessities/lysmarine_gen/tree/bullseye/.circleci

make sure you run parallel make with -j 4
(4 jobs) if you have free CircleCi plan

seandepagnier, 2022-09-03:
I am trying to understand what you are doing and how to reproduce it...

So the detection for wiringpi in the dependency script fails.    How can the detection change to make it work?

mgrouch, 2022-09-03:
To reproduce it you need to try running the build of pypilot on CircleCi instead of your pi. 

right now pypilot is something that’s needed to be built on user’s pi. 

realistically needs to be just a installable package built by one of CI. CircleCi is one of them. 

wiringpi_dep 

function in dependencies.py 

opens

/sys/firmware 

I think it will not work in CI build dockers

mgrouch, 2022-09-03:
Latest BBN Marine OS image (bullseye arm64)
has workarounds for it.

It also should be able to detect Pypilot hat and behave plug n play with it based on our other discussions. 

I have no way to test though without the hardware. 

If you could try latest BBN OS image (bullseye flavor) a test that’s be great. It has fixed for boot speed if you are not network connected too. 

Fair Winds!

seandepagnier, 2022-09-07:
It does open sys/firmware and installs wiring pi if it detects raspberry pi...   Otherwise it doesnt install wiringpi library.

How else can I detect raspberry pi?      Can you add wiring pi dependency outside of pypilot?

mgrouch, 2022-09-08:
I’ve added it externally from pypilot 

https://github.com/bareboat-necessities/lysmarine_gen/blob/bullseye/install-scripts/4-server/02-pypilot.sh

thanks

closed 2022-09-08 by mgrouch