pypilot/pypilot #149: KeyError: 'gps.filtered.output'

mgrouch, 2022-09-07:
```

Sep 06 23:59:11 lysmarine pypilot[67879]: Process Process-4:
Sep 06 23:59:11 lysmarine pypilot[67879]: Traceback (most recent call last):
Sep 06 23:59:11 lysmarine pypilot[67879]:   File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
Sep 06 23:59:11 lysmarine pypilot[67879]:     self.run()
Sep 06 23:59:11 lysmarine pypilot[67879]:   File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
Sep 06 23:59:11 lysmarine pypilot[67879]:     self._target(*self._args, **self._kwargs)
Sep 06 23:59:11 lysmarine pypilot[67879]:   File "/usr/local/lib/python3.9/dist-packages/pypilot/signalk.py", line 257, in process
Sep 06 23:59:11 lysmarine pypilot[67879]:     self.poll(1)
Sep 06 23:59:11 lysmarine pypilot[67879]:   File "/usr/local/lib/python3.9/dist-packages/pypilot/signalk.py", line 311, in poll
Sep 06 23:59:11 lysmarine pypilot[67879]:     self.send_signalk()
Sep 06 23:59:11 lysmarine pypilot[67879]:   File "/usr/local/lib/python3.9/dist-packages/pypilot/signalk.py", line 400, in send_signalk
Sep 06 23:59:11 lysmarine pypilot[67879]:     if sensor == 'gps' and self.last_values['gps.filtered.output'] is True:
Sep 06 23:59:11 lysmarine pypilot[67879]: KeyError: 'gps.filtered.output'
Sep 06 23:59:11 lysmarine pypilot[67879]: signalk process 67879

```

Thanks

seandepagnier, 2022-09-07:
Can you try updating git?

mgrouch, 2022-09-07:
I was up-to-date when I had this issue. I will try updating

seandepagnier, 2022-09-07:
Sorry.    I am trying only to submit fixes at this point and continue development in stable branch.   I realize I failed at this in the past...

closed 2022-09-07 by seandepagnier