pypilot/pypilot #150: ERROR loading wind.py cannot import name 'HeadingOffset'

mgrouch, 2022-09-07:
```

Sep 06 23:59:33 lysmarine pypilot[68001]: world magnetic model not available
Sep 06 23:59:33 lysmarine pypilot[68001]: ERROR loading wind.py cannot import name 'HeadingOffset' from partially initialized module 'pypilot.autopilot' (most likely due to a circular import) (/usr/local/lib/python3.9/dist-packages/pypilot/autopilot.py)   cannot import name 'HeadingOffset' from partially initialized module 'pypilot.autopilot' (most likely due to a circular import) (/usr/local/lib/python3.9/dist-packages/pypilot/autopilot.py)

```

Thanks

mgrouch, 2022-09-07:
I've added wmm2020 to the install procedure for magnetic model. I'll try with it. Not sure if next line error is related to it though

seandepagnier, 2022-09-07:
The wmm is a warning, and only prevent certain minor features from
working but is not essential for the pilot to work.

mgrouch, 2022-09-08:
Circular import is still
present in current version 

thanks

mgrouch, 2022-09-23:
In wind.py

should it be:

`from pypilot.autopilot import resolv`

```

class WindPilot(AutopilotPilot):
  def __init__(self, ap):
    from pypilot.autopilot import HeadingOffset
    super(WindPilot, self).__init__('wind', ap)

```
?