pypilot/pypilot #152: various issues on startup

mgrouch, 2022-09-08:
There are plenty as well as issue ith pypilot dying every 8 sec when connected to signalkwith access token.
Might be this log might help

````
user@lysmarine:~ $ journalctl | grep pypilot
Sep 07 13:29:51 lysmarine systemd[1]: Created slice system-pypilot.slice.
Sep 07 13:29:51 lysmarine systemd[1]: Condition check resulted in pypilot detect being skipped.
Sep 07 13:29:53 lysmarine systemd[1]: Condition check resulted in pypilot detect being skipped.
Sep 07 13:29:55 lysmarine systemd[1]: Started pypilot.
Sep 07 13:29:56 lysmarine systemd[1]: Started pypilot web.
Sep 07 13:30:07 lysmarine pypilot_web[572]: processing clients
Sep 08 04:01:49 lysmarine pypilot[1652]: * Extra verbosity turned on
Sep 08 04:01:49 lysmarine pypilot[1652]: Reading Script: /usr/local/lib/python3.9/dist-packages/wmm2020/setup.cmake
Sep 08 04:01:50 lysmarine pypilot[1652]: -- CMake 3.18.4
Sep 08 04:01:50 lysmarine pypilot[1652]: -- 4 CPU cores detected
Sep 08 04:01:50 lysmarine pypilot[1652]: SetCTestConfiguration:SourceDirectory:/usr/local/lib/python3.9/dist-packages/wmm2020
Sep 08 04:01:50 lysmarine pypilot[1652]: SetCTestConfiguration:BuildDirectory:/usr/local/lib/python3.9/dist-packages/wmm2020/build
Sep 08 04:01:50 lysmarine pypilot[1652]: Run dashboard with model Experimental
Sep 08 04:01:50 lysmarine pypilot[1652]:    Source directory: /usr/local/lib/python3.9/dist-packages/wmm2020
Sep 08 04:01:50 lysmarine pypilot[1652]:    Build directory: /usr/local/lib/python3.9/dist-packages/wmm2020/build
Sep 08 04:01:50 lysmarine pypilot[1652]: CMake Error: Cannot open file for write: /usr/local/lib/python3.9/dist-packages/wmm2020/build/Testing/Temporary/LastStart.log.tmp
Sep 08 04:01:50 lysmarine pypilot[1652]: CMake Error: : System Error: No such file or directory
Sep 08 04:01:50 lysmarine pypilot[1652]: Cannot create log file: LastStart.log
Sep 08 04:01:50 lysmarine pypilot[1652]: CMake Error at /usr/local/lib/python3.9/dist-packages/wmm2020/setup.cmake:60 (ctest_start):
Sep 08 04:01:50 lysmarine pypilot[1652]:   ctest_start unknown error.
Sep 08 04:01:50 lysmarine pypilot[1652]: SetCTestConfiguration:BuildDirectory:/usr/local/lib/python3.9/dist-packages/wmm2020/build
Sep 08 04:01:50 lysmarine pypilot[1652]: SetCTestConfiguration:SourceDirectory:/usr/local/lib/python3.9/dist-packages/wmm2020
Sep 08 04:01:50 lysmarine pypilot[1652]: SetCTestConfiguration:ConfigureCommand:"/usr/bin/cmake" "-GNinja" "/usr/local/lib/python3.9/dist-packages/wmm2020"
Sep 08 04:01:50 lysmarine pypilot[1652]: ctest_configure failed to change directory to /usr/local/lib/python3.9/dist-packages/wmm2020/build : No such file or directory
Sep 08 04:01:50 lysmarine pypilot[1652]: SetCTestConfiguration:BuildDirectory:/usr/local/lib/python3.9/dist-packages/wmm2020/build
Sep 08 04:01:50 lysmarine pypilot[1652]: SetCTestConfiguration:SourceDirectory:/usr/local/lib/python3.9/dist-packages/wmm2020
Sep 08 04:01:50 lysmarine pypilot[1652]: SetCTestConfiguration:ConfigureCommand:"/usr/bin/cmake" "-GNinja" "/usr/local/lib/python3.9/dist-packages/wmm2020"
Sep 08 04:01:50 lysmarine pypilot[1652]: ctest_configure failed to change directory to /usr/local/lib/python3.9/dist-packages/wmm2020/build : No such file or directory
Sep 08 04:01:50 lysmarine pypilot[1652]: -- SKIP: ctest_build(): returncode: ; CMake error code: -1
Sep 08 04:01:50 lysmarine pypilot[1652]: -- SKIP: ctest_test(): returncode: ; CMake error code: -1
Sep 08 04:01:50 lysmarine pypilot[1652]: CMake Error at /usr/local/lib/python3.9/dist-packages/wmm2020/setup.cmake:107 (message):
Sep 08 04:01:50 lysmarine pypilot[1652]:   Build and test failed.
Sep 08 04:01:50 lysmarine pypilot[567]: autopilot start 10.035178529
Sep 08 04:01:50 lysmarine pypilot[567]: world magnetic model not available
Sep 08 04:01:50 lysmarine pypilot[567]: ERROR loading wind.py cannot import name 'HeadingOffset' from partially initialized module 'pypilot.autopilot' (most likely due to a circular import) (/usr/local/lib/python3.9/dist-packages/pypilot/autopilot.py)   cannot import name 'HeadingOffset' from partially initialized module 'pypilot.autopilot' (most likely due to a circular import) (/usr/local/lib/python3.9/dist-packages/pypilot/autopilot.py)
Sep 08 04:01:50 lysmarine pypilot[567]: loading servo calibration /home/pypilot/.pypilot/servocalibration
Sep 08 04:01:50 lysmarine pypilot[567]: WARNING: using default servo calibration!!
Sep 08 04:01:50 lysmarine pypilot[567]: Available Pilots: ['absolute', 'simple', 'basic']
Sep 08 04:01:50 lysmarine pypilot[567]: warning: failed to open special file /dev/watchdog0 for writing
Sep 08 04:01:50 lysmarine pypilot[567]:          cannot stroke the watchdog
Sep 08 04:01:50 lysmarine sudo[1709]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/chrt -pf 2 1707
Sep 08 04:01:50 lysmarine sudo[1706]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/chrt -po 0 1704
Sep 08 04:01:50 lysmarine sudo[1717]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/chrt -pf 1 567
Sep 08 04:01:51 lysmarine pypilot[1707]: Settings file not found. Using defaults and creating settings file
Sep 08 04:01:51 lysmarine pypilot[1707]: Failed to open settings file for saveDetected MPU9250/MPU9255 at standard address
Sep 08 04:01:51 lysmarine pypilot[1707]: Failed to open settings file for saveUsing fusion algorithm Kalman STATE4
Sep 08 04:01:51 lysmarine pypilot[1707]: min/max compass calibration not in use
Sep 08 04:01:51 lysmarine pypilot[1707]: Using ellipsoid compass calibration
Sep 08 04:01:51 lysmarine pypilot[1707]: Using accel calibration
Sep 08 04:01:51 lysmarine sudo[1725]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/chrt -pi 0 1704
Sep 08 04:01:51 lysmarine pypilot[1707]: MPU-925x init complete
Sep 08 04:01:55 lysmarine sudo[1765]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/stty -F /dev/ttyLYS_gps_0
Sep 08 04:04:13 lysmarine signalk-server[992]: PUT /skServer/security/access/requests/pypilot-33041570237/approved 200 14.665 ms - 15
Sep 08 04:04:23 lysmarine pypilot[1712]: Process Process-4:
Sep 08 04:04:23 lysmarine pypilot[1712]: Traceback (most recent call last):
Sep 08 04:04:23 lysmarine pypilot[1712]:   File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
Sep 08 04:04:23 lysmarine pypilot[1712]:     self.run()
Sep 08 04:04:23 lysmarine pypilot[1712]:   File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
Sep 08 04:04:23 lysmarine pypilot[1712]:     self._target(*self._args, **self._kwargs)
Sep 08 04:04:23 lysmarine pypilot[1712]:   File "/usr/local/lib/python3.9/dist-packages/pypilot/signalk.py", line 259, in process
Sep 08 04:04:23 lysmarine pypilot[1712]:     self.poll(1)
Sep 08 04:04:23 lysmarine pypilot[1712]:   File "/usr/local/lib/python3.9/dist-packages/pypilot/signalk.py", line 322, in poll
Sep 08 04:04:23 lysmarine pypilot[1712]:     self.send_signalk()
Sep 08 04:04:23 lysmarine pypilot[1712]:   File "/usr/local/lib/python3.9/dist-packages/pypilot/signalk.py", line 446, in send_signalk
Sep 08 04:04:23 lysmarine pypilot[1712]:     v = self.last_values[key][pypilot_path]
[... 487 lines cut ...]
Sep 08 04:05:11 lysmarine pypilot[4784]: autopilot start 224.513442805
Sep 08 04:05:11 lysmarine pypilot[4784]: world magnetic model not available
Sep 08 04:05:11 lysmarine pypilot[4784]: ERROR loading wind.py cannot import name 'HeadingOffset' from partially initialized module 'pypilot.autopilot' (most likely due to a circular import) (/usr/local/lib/python3.9/dist-packages/pypilot/autopilot.py)   cannot import name 'HeadingOffset' from partially initialized module 'pypilot.autopilot' (most likely due to a circular import) (/usr/local/lib/python3.9/dist-packages/pypilot/autopilot.py)
Sep 08 04:05:11 lysmarine sudo[4797]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/chrt -po 0 4795
Sep 08 04:05:11 lysmarine sudo[4800]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/chrt -pf 2 4798
Sep 08 04:05:11 lysmarine pypilot[4798]: Settings file not found. Using defaults and creating settings file
Sep 08 04:05:11 lysmarine pypilot[4798]: Failed to open settings file for saveDetected MPU9250/MPU9255 at standard address
Sep 08 04:05:11 lysmarine pypilot[4798]: Failed to open settings file for saveUsing fusion algorithm Kalman STATE4
Sep 08 04:05:11 lysmarine pypilot[4798]: min/max compass calibration not in use
Sep 08 04:05:11 lysmarine pypilot[4798]: Using ellipsoid compass calibration
Sep 08 04:05:11 lysmarine pypilot[4798]: Using accel calibration
[... cut ...]
Sep 08 04:05:11 lysmarine pypilot[4784]: Available Pilots: ['absolute', 'simple', 'basic']
Sep 08 04:05:11 lysmarine pypilot[4784]: warning: failed to open special file /dev/watchdog0 for writing
Sep 08 04:05:11 lysmarine pypilot[4784]:          cannot stroke the watchdog
Sep 08 04:05:11 lysmarine sudo[4812]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/chrt -pf 1 4784
Sep 08 04:05:11 lysmarine pypilot[4798]: MPU-925x init complete
Sep 08 04:05:14 lysmarine sudo[4827]:  pypilot : PWD=/ ; USER=root ; COMMAND=/usr/bin/stty -F /dev/ttyLYS_gps_0
user@lysmarine:~ $ 

````

Thanks

seandepagnier, 2022-09-08:
You have this results because you have a gps that does not give track info...   perhaps because you are not moving?
Its different on my gps which did produce track data even though its somewhat random.   I am hoping it is fixed now.

Thanks for this.

mgrouch, 2022-09-08:
opencpn had issue with COG null coming from gps via SignalK Json.
Might be this one is similar type of issue. I'll try with your latest commit

Thanks

mgrouch, 2022-09-08:
Looks like commit 

https://github.com/pypilot/pypilot/commit/9bd72125cdd5b8c50296158f91553ad9802c0823

fixed it

thanks

closed 2022-09-08 by mgrouch